microsoft/Azure-Kinect-Sensor-SDK #313: Azure Pipelines Bot is unable to run only private build

schultetwin1, 2019-05-01:
**Describe the bug**
On Pull Requests, "/azp run Microsoft.Azure-Kinect-Sensor-SDK" should only run the private build. This command does not work.

**To Reproduce**
1. Create a new PR
2. Have an admin comment /azp run Microsoft.Azure-Kinect-Sensor-SDK
3. Nothing happens

schultetwin1, 2019-05-01:
I believe the actual bug here is /azp run Microsoft.Azure-Kinect-Sensor-SDK responds with

> No pipelines are associated with this pull request.

Brent-A, 2019-07-30:
It still seems to repro, but it hasn't been a big issue. @schultetwin1, have you followed up with the pipelines team?

I'm going to close it for now since I don't see us making any updates here.

closed 2019-07-30 by Brent-A