microsoft/Azure-Kinect-Sensor-SDK #324: SDK should detect and gracefully fail on CPUs that do not support SSE 4.1

schultetwin1, 2019-05-03:
**Is your feature request related to a problem? Please describe.**
Currently, if the SDK is run on a CPU that does not support SSE 4.1 instructions, the code will fault due to an unrecognized instructions.

**Describe the solution you'd like**
A cross platform library to run time detect SSE 4.1 support on Linux and Windows. This detection will gate running of the SDK.

Brent-A, 2019-05-07:
Before approving we want to have a good idea of how the check will be done, and what we need to do to verify it.

diablodale, 2019-11-09:
The app itself (or the SDK) can do the CPU check with libraries like https://github.com/google/cpu_features
There are other libraries if you don't prefer that specific licenser.