microsoft/Azure-Kinect-Sensor-SDK #328: Current SDK (1.0.1 and also 1.0.0) package cannot be compiled in Visual Studio 2019 on my computer.

afantima, 2019-05-05:
**Describe the bug**
I recently upgrade Visual Studio software from 2017 to 2019. Before upgrading, I can compile under 2017 version, but after upgrading to 2019, it cannot. It will stop on  CMake generation and cannot move forward.

**To Reproduce**
On my computer.
1) Run GitHub Desktop and pull in latest SDK package (5/5/2019).
2) After pulling in request has been finished, run Visual Studio 2019 and open CMakeLists.txt to open this project (by File-->Open-->Cmake)
3) Cmake IntelliSense will be stopped because of VS2019.

**Expected behavior**
Code can be compiled out of box.

**Logs**
Here is the failing information on VS2019.
1> CMake generation started.
1> Command line: c:\program files (x86)\microsoft visual studio\2019\professional\common7\ide\commonextensions\microsoft\cmake\CMake\bin\cmake.exe  -G "Visual Studio 15 2017 Win64" -DCMAKE_INSTALL_PREFIX:PATH="C:\OS\Azure-Kinect-Sensor-SDK\install\Win-x64-Release-MSBuild" -DCMAKE_CXX_COMPILER:FILEPATH="C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/VC/Tools/MSVC/14.20.27508/bin/HostX86/x64/cl.exe" -DCMAKE_C_COMPILER:FILEPATH="C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/VC/Tools/MSVC/14.20.27508/bin/HostX86/x64/cl.exe"  -DCMAKE_CONFIGURATION_TYPES="RelWithDebInfo" -DCMAKE_GENERATOR_INSTANCE="C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional" "C:\OS\Azure-Kinect-Sensor-SDK"
1> Working directory: C:\OS\Azure-Kinect-Sensor-SDK\build\Win-x64-Release-MSBuild
1> [CMake] -- Configuring incomplete, errors occurred!
1> [CMake] See also "C:/OS/Azure-Kinect-Sensor-SDK/build/Win-x64-Release-MSBuild/CMakeFiles/CMakeOutput.log".
1> [CMake] CMake Error at C:\OS\Azure-Kinect-Sensor-SDK\CMakeLists.txt:23 (project):
1> [CMake]   Failed to run MSBuild command:
1> [CMake] 
1> [CMake]     MSBuild.exe
1> [CMake] 
1> [CMake]   to get the value of VCTargetsPath:
1> [CMake] 
1> [CMake]     Microsoft (R) Build Engine version 16.0.461+g6ff56ef63c for .NET Framework
1> [CMake]     Copyright (C) Microsoft Corporation. All rights reserved.
1> [CMake]     
1> [CMake]     Build started 5/5/2019 10:56:25 AM.
1> [CMake]     Project "C:\OS\Azure-Kinect-Sensor-SDK\build\Win-x64-Release-MSBuild\CMakeFiles\3.13.19031502-MSVC_2\VCTargetsPath.vcxproj" on node 1 (default targets).
1> [CMake]     C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppBuild.targets(378,5): error MSB8020: The build tools for Visual Studio 2017 (Platform Toolset = 'v141') cannot be found. To build using the v141 build tools, please install Visual Studio 2017 build tools.  Alternatively, you may upgrade to the current Visual Studio tools by selecting the Project menu or right-click the solution, and then selecting "Retarget solution". [C:\OS\Azure-Kinect-Sensor-SDK\build\Win-x64-Release-MSBuild\CMakeFiles\3.13.19031502-MSVC_2\VCTargetsPath.vcxproj]
1> [CMake]     Done Building Project "C:\OS\Azure-Kinect-Sensor-SDK\build\Win-x64-Release-MSBuild\CMakeFiles\3.13.19031502-MSVC_2\VCTargetsPath.vcxproj" (default targets) -- FAILED.
1> [CMake]     
1> [CMake]     Build FAILED.
1> [CMake]     
1> [CMake]     "C:\OS\Azure-Kinect-Sensor-SDK\build\Win-x64-Release-MSBuild\CMakeFiles\3.13.19031502-MSVC_2\VCTargetsPath.vcxproj" (default target) (1) ->
1> [CMake]     (PrepareForBuild target) -> 
1> [CMake]       C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppBuild.targets(378,5): error MSB8020: The build tools for Visual Studio 2017 (Platform Toolset = 'v141') cannot be found. To build using the v141 build tools, please install Visual Studio 2017 build tools.  Alternatively, you may upgrade to the current Visual Studio tools by selecting the Project menu or right-click the solution, and then selecting "Retarget solution". [C:\OS\Azure-Kinect-Sensor-SDK\build\Win-x64-Release-MSBuild\CMakeFiles\3.13.19031502-MSVC_2\VCTargetsPath.vcxproj]
1> [CMake]     
1> [CMake]         0 Warning(s)
1> [CMake]         1 Error(s)
1> [CMake]     
1> [CMake]     Time Elapsed 00:00:00.28
1> [CMake]     
1> [CMake] 
1> [CMake]   Exit code: 1
1> [CMake] 
1> [CMake] 
1> [CMake] 
1> [CMake] 
1> [CMake] 
1> CMake generation finished.
1> c:\program files (x86)\microsoft visual studio\2019\professional\common7\ide\commonextensions\microsoft\cmake\CMake\bin\cmake.exe  -G "Visual Studio 15 2017 Win64" -DCMAKE_INSTALL_PREFIX:PATH="C:\OS\Azure-Kinect-Sensor-SDK\install\Win-x64-Release-MSBuild" -DCMAKE_CXX_COMPILER:FILEPATH="C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/VC/Tools/MSVC/14.20.27508/bin/HostX86/x64/cl.exe" -DCMAKE_C_COMPILER:FILEPATH="C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/VC/Tools/MSVC/14.20.27508/bin/HostX86/x64/cl.exe"  -DCMAKE_CONFIGURATION_TYPES="RelWithDebInfo" -DCMAKE_GENERATOR_INSTANCE="C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional" "C:\OS\Azure-Kinect-Sensor-SDK" returned with exit code: 1

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS with Version: <1809>
 - SDK Version:  <1.0.1>

Brent-A, 2019-05-05:
It looks like our Cmake Settings.json file in the root of the project is configured for the vs2017 tool chain for the *msbuild project configurations. You might be able to just use the drop down at the top of visual studio to switch for win-x64-debug-msbuild to win-x64-debug-ninja. 

I will have to try this tomorrow and see if there is a way to switch the version depending on what version of visual studio is installed.

schultetwin1, 2019-05-06:
We are currently not testing VS2019 MSBuild in our CI system, only VS2019 Ninja. As Brent-A said, switching from Win-x64-Release-MSBuiild to Win-x64-Release-Ninja should work. I have tested this on my machine but let me know if it works for you.

@Brent-A it looks like the VS2019 generator will not be supported in CMakeSettings.json until a later time.
https://developercommunity.visualstudio.com/comments/499124/view.html

Brent-A, 2019-05-07:
@afantima were you able to build the project with the Ninja configuration?

Brent-A, 2019-05-07:
We will try to get a repro with the same tool set. I was unable to load the CMake project at all on my 2019 preview install, so we need to narrow this down to problems with VS2019 and our configuration.

Brent-A, 2019-07-30:
Once 2019 is more widely adopted and has better cmake we will move the .json file up. Until then, VS2017 or ninja configurations are your best bet.

closed 2019-07-30 by Brent-A