microsoft/Azure-Kinect-Sensor-SDK #329: ARM (Linux) architecture support

szi, 2019-05-07:
**Is your feature request related to a problem? Please describe.**
Currently I cannot use the sensor with ARM/Linux-based devices. Many embedded/robotics targets run on ARM. For example, it would be nice if you'd support NVIDIA Jetson Nano or similar embedded dev platforms.

**Describe the solution you'd like**
It sounds like the only thing needed would be a precompiled version of the DepthEngine closed-source binary for a generic ARM target? Ideally provide versions for both hardfp and softfp ABI.

Brent-A, 2019-05-07:
Thanks for the suggestion.

We want to consider this for a future release. We need more feedback from customers on which specific platforms they are interested in us supporting so that we can invest in the right build and test infrastructure to enable those platforms.

Brent-A, 2019-05-10:
@szi, I just posted your suggestion to our new [feedback forum](https://feedback.azure.com/forums/920053-azure-kinect-dk). Go ahead and up vote it there to show your support for the feature, or comment if you want to provide more details.

closed 2019-05-10 by Brent-A

37, 2019-06-11:
:+1: for ARM support. Specifically, I'd love to be able to run it on a Jetson TX2.

zengh100, 2019-07-11:
@Brent-A, any update since this topic is closed on May 10?
One question about https://packages.microsoft.com/ubuntu/16.04/multiarch/prod/dists/xenial/main/binary-arm64/: 
It seemed that this file https://packages.microsoft.com/config/ubuntu/16.04/multiarch/arm64/prod.list is wrong about the package location as below:
deb [arch=arm64] https://packages.microsoft.com/ubuntu/16.04/multiarch/arm64/prod xenial main
So I changed it to:
deb [arch=arm64] https://packages.microsoft.com/ubuntu/16.04/multiarch/prod xenial main