microsoft/Azure-Kinect-Sensor-SDK #332: A small number of devices fail to read the ContainerID

wes-b, 2019-05-08:
**Describe the bug**
A small number of devices work fine when WinUsb is used to communicate with the device. However when using with LibUsb and reading the Container ID, a timeout reading the BOS descriptor occurs.

**To Reproduce**
1. Attach device to PC
2. Run any test that calls k4a_device_open()

**Expected behavior**
k4a_device_open() should succeed, but it blocks for about 1 minute and then fails.

**Logs**
N/A

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS with Version: 1903
 - SDK Version:  All
 - Firmware version:  9870

**Additional context**
N/A

joylital, 2019-05-08:
Devices with this failure won't pass manufacturing line and fail first in device calibration.

joylital, 2019-05-28:
There is a fix in FW (Depth ISP 1.6.74), to be validated.

wes-b, 2019-06-03:
This turned out to be a firmware issue

closed 2019-06-03 by wes-b