microsoft/Azure-Kinect-Sensor-SDK #335: UWP Support

Noemata, 2019-05-08:
Will this SDK ever support UWP, and if not, why?

joylital, 2019-05-10:
Thanks for the request @Noemata , we do not have plans to support UWP today. Last //build conference we started IWANTKINECT survey based on those results we prioritized cross-platform SDK that works with Win32 and Linux. 

We are working on getting feedback system available that can inform us better on overall community interests/needs so we can continue prioritizing development. I will post a link here as soon as that is available.

Noemata, 2019-05-10:
The waffling on platform support initiatives have been painful to both watch and endure over the years.  All of a sudden, UWP gets the bottom rung.  WPF suffered a similar demotion when UWP first game out.  And now Microsoft is somehow intent on resurrecting Win32 anew??  The Silverlight saga should have been a pretty good object lesson.  If you want to commit to a change, then go all the way.  Which is it now, WPF, Winforms, UWP, DirectX + Win32 (sarcasm)?  My point being, UWP support cannot slip.  There will be no third chance since this time around we'll be at strike three.

Brent-A, 2019-05-10:
Hi @Noemata. Thanks for your feedback. 

We're are collecting community suggestions on the future of this project at our new [user feedback forum](https://feedback.azure.com/forums/920053). I just added your request there, and you can upvote it to voice your support.

The Azure Kinect DK is still a developer kit and we are looking for specific ways that you want to use the device. Please comment on the [user feedback forum](https://feedback.azure.com/forums/920053) since this will help us make decisions about what platforms and features will be most useful going forward.

closed 2019-05-10 by Brent-A

Noemata, 2019-06-28:
Curiously, UWP support is one of the more requested items at the user feedback forum.  Has anyone noticed?  Or is there a vote threshold required before things get attention.  I can arrange for an entire college student body that needs this to put in a vote.