microsoft/Azure-Kinect-Sensor-SDK #340: Linux - Can't find UVC device

jeffdelmerico, 2019-05-13:
Built SDK from develop branch on Ubuntu 16.04, and udev rules have been installed to enable non-root usage.

Device works as expected with Windows SDK,  On linux, the device is detected and enumerated with `lsusb`, but produces the following errors when running `enumerate_devices`:
```
Found 1 connected devices:
[2019-05-13 13:32:47.162] [error] [t=11166] ../src/color/uvc_camerareader.cpp (53): Init(). Can't find UVC device: No such device
[2019-05-13 13:32:47.162] [error] [t=11166] ../src/color/color.cpp (83): color->m_spCameraReader->Init(serial_number) returned failure in color_create()
[2019-05-13 13:32:47.162] [error] [t=11166] ../src/sdk/k4a.c (173): color_create( device->tick_handle, container_id, serial_number, color_capture_ready, handle, &device->color) returned failure in k4a_device_open()
0: Failed to open device
0: Failed to get serial number length
```
with similar output when running `k4aviewer`:
```
[2019-05-13 13:26:15.071] [error] [t=11035] ../src/color/uvc_camerareader.cpp (53): Init(). Can't find UVC device: No such device
[2019-05-13 13:26:15.071] [error] [t=11035] ../src/color/color.cpp (83): color->m_spCameraReader->Init(serial_number) returned failure in color_create()
[2019-05-13 13:26:15.071] [error] [t=11035] ../src/sdk/k4a.c (173): color_create( device->tick_handle, container_id, serial_number, color_capture_ready, handle, &device->color) returned failure in k4a_device_open()
```

The output in `dmesg` indicates some issue with UVC, but information about this error is scarce.
```
[243073.455732] usb 1-1: new high-speed USB device number 21 using xhci_hcd
[243073.585573] usb 1-1: descriptor type invalid, skip
[243073.588098] usb 1-1: New USB device found, idVendor=045e, idProduct=097b
[243073.588123] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[243073.588143] usb 1-1: Product: 4-Port USB 2.0 Hub
[243073.588155] usb 1-1: Manufacturer: Generic
[243073.588158] usb 1-1: SerialNumber: 20181115115945
[243073.589316] hub 1-1:1.0: USB hub found
[243073.589584] hub 1-1:1.0: 4 ports detected
[243073.695795] usb 2-1: new SuperSpeed USB device number 36 using xhci_hcd
[243073.714597] usb 2-1: New USB device found, idVendor=045e, idProduct=097a
[243073.714619] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[243073.714622] usb 2-1: Product: 4-Port USB 3.0 Hub
[243073.714626] usb 2-1: Manufacturer: Generic
[243073.714629] usb 2-1: SerialNumber: 20181115115945
[243073.716944] hub 2-1:1.0: USB hub found
[243073.717193] hub 2-1:1.0: 2 ports detected
[243074.263561] usb 1-1.3: new high-speed USB device number 22 using xhci_hcd
[243074.356452] usb 1-1.3: New USB device found, idVendor=045e, idProduct=097e
[243074.356459] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[243074.356463] usb 1-1.3: Product: Azure Kinect Microphone Array
[243074.356466] usb 1-1.3: Manufacturer: Microsoft Corporation.
[243074.356469] usb 1-1.3: SerialNumber: 000235484312
[243074.360536] hid-generic 0003:045E:097E.000A: hiddev0,hidraw1: USB HID v1.11 Device [Microsoft Corporation. Azure Kinect Microphone Array] on usb-0000:00:14.0-1.3/input2
[243074.427682] usb 2-1.1: new SuperSpeed USB device number 37 using xhci_hcd
[243074.444133] usb 2-1.1: Isoc endpoint with wBytesPerInterval of 1024 in config 1 interface 1 altsetting 4 ep 129: setting to 944
[243074.444486] usb 2-1.1: New USB device found, idVendor=045e, idProduct=097d
[243074.444490] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[243074.444494] usb 2-1.1: Product: Azure Kinect 4K Camera
[243074.444497] usb 2-1.1: Manufacturer: Microsoft
[243074.444500] usb 2-1.1: SerialNumber: 00023548412
[243074.445843] uvcvideo: Found UVC 1.00 device Azure Kinect 4K Camera (045e:097d)
[243074.447046] uvcvideo: Failed to set UVC probe control : -32 (exp. 26).
[243074.450493] input: Azure Kinect 4K Camera as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.1/2-1.1:1.0/input/input25
[243074.523544] usb 2-1.2: new SuperSpeed USB device number 38 using xhci_hcd
[243074.540158] usb 2-1.2: New USB device found, idVendor=045e, idProduct=097c
[243074.540163] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[243074.540167] usb 2-1.2: Product: Azure Kinect Depth Camera
[243074.540170] usb 2-1.2: Manufacturer: Microsoft
[243074.540173] usb 2-1.2: SerialNumber: 000235484312
[243146.024456] usb 2-1.1: USB disconnect, device number 37
[243146.188609] usb 2-1.2: USB disconnect, device number 38
[243147.075471] usb 1-1: USB disconnect, device number 21
[243147.075480] usb 1-1.3: USB disconnect, device number 22
[243149.305891] usb 2-1: USB disconnect, device number 36
```

wes-b, 2019-05-13:
Thanks @jeffdelmerico, we will try to repro and address shortly.

jeffdelmerico, 2019-05-15:
Thanks. Please let me know if I can provide any other information to support.

jeffdelmerico, 2019-05-15:
I have an update...I was able to test my installation with a different device, and this time it was successful.  I'm not sure if it's relevant, but with the working device, there is a small difference in the output of AzureKinectFirmwareTool:
```
Current Firmware Versions:
  RGB camera firmware:      1.5.92
  Depth camera firmware:    1.5.66
  Depth config file:        5006.27
  Audio firmware:           1.5.14
  Build Config:             Production
  Certificate Type:         Microsoft
```
Whereas the the non-functioning one has the same firmware versions, but the certificate type is "Unsigned".  Otherwise, as far as I can tell, the two devices should be identical.

wes-b, 2019-05-15:
Thanks @jeffdelmerico, you have a preproduction unit, so lets continue this conversation offline.

wes-b, 2019-05-17:
Opened #358 to track the unclear error message. 

This issue is caused by the depth camera and color camera having different serial numbers. They should be the same for production hardware

closed 2019-05-17 by wes-b