microsoft/Azure-Kinect-Sensor-SDK #357: fastpointcloud example ply missing mesh face triangles

dimitrijevic, 2019-05-17:
The previous Kinect 2.0 SDK provided us with the ability to export a mesh beside the point cloud, which seems to be currently missing in the K4A SDK, and as such will make us write more code which the SDK might solve.

If possible please provide the ability to generate a vertex colored mesh as well as point cloud.

If this is not an SDK issue, could you please provide some guidance on what alternative is best used, such as KinFu (or is there any other custom solutions for generating a TSDF cloud and a subsequent mesh)?

Thanks in advance, either way, really.

rabbitdaxi, 2019-05-29:
Hi dimitrijevic, thank you for the feedback. Yes, we will look into this feature request. Beside of that, we can provide an example that leveraging the Azure Kinect Sensor SDK and OpenCV kinfu module (from contribute), it should give you live kinect fusion TSDF surface render as well as fused point cloud output.

dimitrijevic, 2019-05-30:
Awesome, thank you, I was contemplating, using PCL myself maybe, when it hit 1.10 hopefully...but it says their implementation's outdated, so OpenCV is an even better choice...I was hoping maybe to see Microsoft's implementation though?

Once this and an example repo is available I'll try and make a simple mex call of the C++ function that provides the mesh so the new Kinect can be used inside MATLAB as well like with the Kin2 (although that was missing colored mesh which I'm hoping the OpenCV implementation has) I'll look into it myself later this week maybe.

TIA all the same rabbutdaxi&cdedmonds

dimitrijevic, 2019-06-17:
As promised ([some shameless self-promo](http://github.com/dimitrijevic/k4a3d)) (very simple MATLAB mex wrapper).
I know this is quite simple code but it may be of use to someone starting with MATLAB just as myself recently... 
Now to adapt the KinFu example I'd like to ask first how much do you guys plan to change the SDK, most importantly will the separate procedures for device and recording playback be unified and as such does it have a timeline, since that would make any MATLAB example deprecated IMHO, also will OpenCV be the recommended CV library or will you be rolling those processing features into the SDK in the future? TIA

rabbitdaxi, 2019-06-26:
Just back from a conference, two follow up on this:
1) we published the sample that using Azure Kinect SDK with the OpenCV KinFu in our sample repo, which will be public soon
2) we have opened a feedback user voice to track the this feature request, https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/37981348-kinect-fusion-support

dimitrijevic, 2019-06-27:
No problem @rabbitdaxi, I kinda found/downloaded it, prior among pull requests already...thanks though....however, I haven't had the chance yet, to look at it and make mexFuncrion. As soon as I replace old SSD, and build the needed OpenCV contrib from sources, I'll be looking into it some more I promise. 

As for the suggestions, just the one, for now...maybe, if possible, make a tool, for simulating the device, but that probably requires services, and the new SDK being multiplatform, I'm not sure if Kinect Studio's possible...that way no playback's needed, decreasing the number of methods, should I put that on UserVoice too?

rabbitdaxi, 2019-06-27:
Hi @dimitrijevic , thank you for the suggestion, of course, any feedback is great, UserVoice is the place you can share feedback.

rabbitdaxi, 2019-06-27:
Close the github issue based on the fact that we have a sample for opencv kinfu as one approach and we have the user voice to track a long term plan on Kinect Fusion support

closed 2019-06-27 by rabbitdaxi