microsoft/Azure-Kinect-Sensor-SDK #363: Deadlock in transform functions

Helco, 2019-05-22:
**Describe the bug**

The transform functions (e.g. `k4a_transformation_color_image_to_depth_camera`) are susceptible to deadlocks (I would guess by lost wakeup). These won't occur in the example because the functions is only called once there.

**To Reproduce**

Call some transformation functions in a loop and wait. The lockdown occurred much faster on my slower devices. I assembled a minimal repro application [here](https://gist.github.com/Helco/00d2adc0147e17237a1e7155ababeb2a)

**Expected behavior**

An infinite loop of transformed frames.

**Desktop (please complete the following information):**
 - Windows 1903
 - c2949cc06c82530d153f33ca450f09303fc0c22a

mbleyer, 2019-05-22:
Not sure if I understand the bug. Does your example program hang or does it run out of memory? It seems like the images are not deallocated.

Helco, 2019-05-23:
The program hangs between getting the frame and processing it. Thanks to C++ RAII the image gets deallocated after line 37.

Brent-A, 2019-05-23:
Do you know if you still reproduce the problem if you don't re-create the transformation object in each iteration of the loop?

mbleyer, 2019-05-23:
I was able to repro using Helco's code sample. The issue goes away when initializing the transformation object outside of the while loop, which represents the right use of the SDK. We might want to highlight that in the documentation. 

@christianmakela: We can try investigating why the deadlock occurs. It would be expected that the frame rate drops significantly in Helco's code, but we should not see a deadlock.

Brent-A, 2019-05-23:
Yes, we should probably highlight that the transformation object itself is somewhat heavy weight. I can see why it wouldn't be obvious though. The C API hints at this, but I imagine that C++ consumers won't dig that deep unless they have a problem.

In any case, we certainly shouldn't deadlock.

mbleyer, 2019-05-23:
I tried a few more times and was not able to repro the deadlock anymore. Andrew @rabbitdaxi is looking at it now.

Helco, 2019-05-24:
Great stuff, with the workaround I also couldn't reproduce the deadlock on my machine. Also as mbleyer guessed the frame rate improved quite a lot.

rabbitdaxi, 2019-05-24:
Meanwhile, I will take a deeper look with a low end computer, it might be a race of the worker thread and main thread mutex was not robust enough to prevent the dead lock when user loop the transformation_create (although which is not the recommened way), however, we still need to fix the dead lock if we find a way to repro :)

PR #376 (2019-05-24): Ensure the worker thread loop to acquire the worker lock first

## Fixes #363 

### Description of the changes:
- Ensure the worker thread loop to acquire the worker lock before the process frame function in main thread acquires the worker lock to avoid dead lock

### Before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [x] I [built my changes](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) locally
- [x] I ran the [unit tests](https://github.com/Microsoft/Az [... cut]
closed 2019-05-24 by rabbitdaxi