microsoft/Azure-Kinect-Sensor-SDK #366: Triggering Azure Kinects externally using 3.5mm audio cable (using RPi etc)

this-username-is-taken, 2019-05-22:
**Is your feature request related to a problem? Please describe.**

I'd like to have an external device trigger Azure Kinects and control the triggering time. For example, extend an 3.5mm audio cable from a RPi or Arduino and connect to Azure Kinects (all in slave mode). Essentially, the RPi/Arduino will act as master and trigger the Azure Kinects.

**Describe the solution you'd like**

I'd like to know what the triggering sequence is. In other words, what signals I should send through the 3.5mm audio cable in order to trigger an Azure Kinect capture.

wes-b, 2019-05-22:
Good idea, I need to gather info about this.

tesych, 2019-07-15:
1.	The SYNC signals are active high and pulse width should be greater than 8us.
2.	We currently support 30, 15 and 5 fps, so frequency of master VSYNC signal from your board (developed to generate signal) should also be one of these 3.
3.	Kinect DK 1 should be connected  to your board, Kinect DK 2 to your Kinect DK 1 and Kinect DK 3 to your Kinect DK 2, say daisy-chain.
4.	SYNC signal from the board should be 5V TTL/CMOS with maximum driving capacity no less than 8mA.
5.	The below plugs can be used with Kinect DK, and all sleeves and rings are shorted together inside Kinect DK and they are connected to ground of master Kinect DK. Tip is the sync signal.
 
[image]

tim-depthkit, 2019-07-16:
Would the Azure Kinect respond poorly to a sync signal produced at a broadcast frame rate, such as 29.97?

TomHVR, 2019-09-20:
Maybe look into this hardware to convert the signals?

https://www.canford.co.uk/Products/90-3029_LYNX-YELLOBRIK-SPG-1707-SYNC-PULSE-GENERATOR-3x-HD-tri-level-3x-SD-bi-level-genlock-input

wes-b, 2019-10-02:
@tesych is this done?

jmarkow, 2019-11-13:
Hopping on this thread for something related.  I'd like to sync external equipment with the Kinect.  If I activate the SYNC OUT port on a Kinect, what signal is sent out exactly (pulse width, voltage, etc.) and how does that signal relate to the timing of each frame?

wes-b, 2019-11-14:
The signal is the subordinates VSYNC. You can use subordinate_delay_off_master_usec set a delay or deliver the pulse immediately.

neurojaym, 2019-11-26:
Hey all, jumping onboard with some more information and questions regarding external device control of a Kinect Azure (all on Windows 10 Pro, SDK v1.3.0). I've been trying to control the K4 with an Arduino Mega2560 microcontroller. I'm sending out a fixed number of 1ms, 30Hz, >5V pulses from the arduino to the sync-in port on the K4 which is designated as subordinate. 
Workflow is as follows: 
1. Go to command line and type `k4recorded.exe -l 5 --external-sync sub <filepath>` for a 5 sec recording
2. The cmd line prompt shows: device info, `device started...[subordinate mode] Waiting for signal from master...`
3. Program stalls and waits
4. I hit the start/reset button on my arduino microcontroller which begins 30Hz pulsing for 5s (verified by LED)
5. Recording automatically stops on k4 and saves the correct 5s/150frame .mkv file

However, I tried to break my own system as a check for robustness and noticed the following: if I designate the k4recorder file to be longer, say 10 seconds [`-l 10` in cmd line], while maintaining a 5s pulse train from the arduino, I notice that the recording still starts at the designated time (when I start the Arduino) but records a full 10 seconds, despite only 5 seconds of Arduino-generated pulses, suggesting that subordinate k4's don't actually listen to individual pulses coming in from an external "master", rather they rely purely on their own hardware to record and only "listen" to the opening pulse to timelock signals across daisy-chained components. @microsoft, is this a correct interpretation of what might be happening here? Could you please provide some more insight regarding how subordinate k4's read in information from the 'sync in' port? Thanks very much!

wes-b, 2019-11-27:
Correct, while the subordinate will wait for the first pulse from master, the subordinate no longer needs to pulse to take samples N+1 and beyond. The pulses do need to stay periodic and matched as that is how the 90KHz clocks between the two devices are resynchronized to help avoid drift.

tesych, 2020-01-19:
documented in [Multi-camera](https://docs.microsoft.com/en-us/azure/Kinect-dk/multi-camera-sync) sync article.

closed 2020-01-19 by tesych

sbrahma0, 2022-06-01:
I am able to hook up 2 kinects and trigger to start recording, I want to trigger to stop recording as well. Right now I have to either have a timer or manually press ctrl+z to stop recording. Are there ways to externally trigger to stop recording, if yes plese direct me to those examples. Thanks

histun, 2022-08-11:
Same as above. Trigger start works fine, but I'm wondering if there is a trigger stop as well.

Johunt-feng, 2022-08-18:
Same as above. When  the signal generator to stop the external trigger signal, the device did not stop capturing images.

sbrahma0, 2023-03-22:
Same as above, Trigger helps to start the recording, but is there a way to stop the recording using external signal. Regarding the external trigger, is there a way to start the kernal and trigger the recording with some internal events (like "key board" or message received form some other source) other than external signal?