microsoft/Azure-Kinect-Sensor-SDK #370: Include CMake and MS Build files in the MSI

MsDerekMa, 2019-05-23:
**Is your feature request related to a problem? Please describe.**
The MSI currently contains the DLLs, Headers, and Libs required to build an application against the SDK. However there are not CMake or Build target files included to simplify referencing the 

**Describe the solution you'd like**
We are already generating CMake target files with our build, in addition we are creating a NuGet package which requires MSBuild target files.  These files should be included in the MSI to make it easier to build against the SDK.

MsDerekMa, 2019-05-23:
We should also evaluate adding the CMake files to the NuGet package as well.

UnaNancyOwen, 2019-05-30:
I agree with this proposal.
Please pack the directory (<code>C:\Program Files\K4A</code>) where you built and installed the Azure Kinect SDK using CMake as it is. It contains all the necessary files to generate user project using CMake.
[image]

UnaNancyOwen, 2019-07-19:
If possible, Please contain CMake files in the Azure Kinect Body Tracking SDK too.
We want to handle the Azure Kinect Body Tracking SDK with CMake as well.
It can be written by the user. However, It is no good for users to write find module.
If possible, It should be provided by officially.

mfoco, 2020-08-10:
Any update on this? It would really be convenient to get these files, instead of having to copy them around from somewhere.

nvibd, 2021-02-12:
Bump, I am also still interested in this.