microsoft/Azure-Kinect-Sensor-SDK #374: Putting a Linux system to seal while streaming make the device non-responsive

MsDerekMa, 2019-05-24:
**Describe the bug**
When putting a Linux computer to sleep while streaming, the device doesn't appear to stop streaming. When the computer is woken back up, the LED is still on and the K4A Viewer errored out. You can't start a new stream. Only way to recover is to unplug and re-plug the device.

**To Reproduce**
1. Open up K4A Viewer
2. Start streaming all sensor data
3. Lock the Linux system, and then put the system to sleep
4. Wake the system back up
5. Notice that K4A Viewer has failed. Short of unplugging the device, there is no way to start streaming again.

**Expected behavior**
When the computer wakes back up, it is possible to start streaming again.

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04.2 LTS
 - SDK Version:  1.1.0-beta.0

**Logs**
[ info     ] : find_libusb_device(). Container ID found: {55790093-fbf5-4587-3c9d-408ca307681e} 
[ info     ] : populate_serialnumber(). Serial Number found 000090590812
[ info     ] : find_libusb_device(). Container ID found: {55790093-fbf5-4587-3c9d-408ca307681e} 
[ info     ] : populate_serialnumber(). Serial Number found 000090590812
[ info     ] : ******************** Device Info ********************
[ info     ] : K4A SDK version:     1.1.0-beta.0
[ info     ] : Serial Number:       000090590812
[ info     ] : RGB Sensor Version:  1.6.98
[ info     ] : Depth Sensor Version:1.6.70
[ info     ] : Mic Array Version:   1.6.14
[ info     ] : Sensor Config:       6109.7
[ info     ] : Build type:          Release
[ info     ] : Signature type:      MSFT
[ info     ] : ****************************************************
[ info     ] : k4a_device_start_cameras(). Starting camera's with the following config.
[ info     ] : k4a_device_start_cameras().     color_format:3
[ info     ] : k4a_device_start_cameras().     color_resolution:1
[ info     ] : k4a_device_start_cameras().     depth_mode:2
[ info     ] : k4a_device_start_cameras().     camera_fps:2
[ info     ] : k4a_device_start_cameras().     synchronized_images_only:1
[ info     ] : k4a_device_start_cameras().     depth_delay_off_color_usec:0
[ info     ] : k4a_device_start_cameras().     wired_sync_mode:0
[ info     ] : k4a_device_start_cameras().     subordinate_delay_off_master_usec:0
[ info     ] : k4a_device_start_cameras().     disable_streaming_indicator:0
[ info     ] : verify_plugin(). Loaded K4A Plugin with version: 1.0.7
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  88033333
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  88066666
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  88099988
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  88133333
[ warning  ] : depth_engine_thread(). Dropping depth image due to bad timestamp at startup
[ info     ] : k4a_device_start_cameras(). k4a_device_start_cameras started
[ info     ] : k4a_device_start_imu(). k4a_device_start_imu started
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[ error    ] : usb_cmd_libusb_cb(). Error LIBUSB transfer failed, result:LIBUSB_TRANSFER_NO_DEVICE
[ warning  ] : imu_capture_ready(). A streaming IMU transfer failed
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : queue_disable(). Waiting for blocking call to complete
[ error    ] : imu_get_sample(device->imu, imu_sample, timeout_in_ms) returned failure in k4a_device_get_imu_sample()
[ info     ] : k4a_device_stop_imu(). k4a_device_stop_imu stopping
[ info     ] : k4a_device_stop_cameras(). k4a_device_stop_cameras stopping
[ error    ] : cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[ error    ] : depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
[ warning  ] : queue_disable(). Waiting for blocking call to complete
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : capturesync_add_capture(). Capture Error Detected, Depth 
[ info     ] : k4a_device_stop_cameras(). k4a_device_stop_cameras stopped
[ info     ] : find_libusb_device(). Container ID found: {55790093-fbf5-4587-3c9d-408ca307681e} 
[ info     ] : populate_serialnumber(). Serial Number found 000090590812
[ info     ] : find_libusb_device(). Container ID found: {55790093-fbf5-4587-3c9d-408ca307681e} 
[ info     ] : populate_serialnumber(). Serial Number found 000090590812
[ info     ] : ******************** Device Info ********************
[ info     ] : K4A SDK version:     1.1.0-beta.0
[ info     ] : Serial Number:       000090590812
[ info     ] : RGB Sensor Version:  1.6.98
[ info     ] : Depth Sensor Version:1.6.70
[ info     ] : Mic Array Version:   1.6.14
[ info     ] : Sensor Config:       6109.7
[ info     ] : Build type:          Release
[ info     ] : Signature type:      MSFT
[ info     ] : ****************************************************
[ info     ] : k4a_device_start_cameras(). Starting camera's with the following config.
[ info     ] : k4a_device_start_cameras().     color_format:3
[ info     ] : k4a_device_start_cameras().     color_resolution:1
[ info     ] : k4a_device_start_cameras().     depth_mode:2
[ info     ] : k4a_device_start_cameras().     camera_fps:2
[ info     ] : k4a_device_start_cameras().     synchronized_images_only:1
[ info     ] : k4a_device_start_cameras().     depth_delay_off_color_usec:0
[ info     ] : k4a_device_start_cameras().     wired_sync_mode:0
[ info     ] : k4a_device_start_cameras().     subordinate_delay_off_master_usec:0
[ info     ] : k4a_device_start_cameras().     disable_streaming_indicator:0
[ error    ] : usb_cmd_write(). Write command(80000001) ended in failure, Command status 0x00000001
[ error    ] : usb_cmd_write( colormcu->usb_cmd, DEV_CMD_SET_SYS_CFG, (uint8_t *)&sync_config, sizeof(sync_config), NULL, 0) returned failure in colormcu_set_multi_device_mode()
[ error    ] : colormcu_set_multi_device_mode(device->colormcu, config) returned failure in k4a_device_start_cameras()
[ info     ] : k4a_device_start_cameras(). k4a_device_start_cameras started
[ info     ] : k4a_device_stop_cameras(). k4a_device_stop_cameras stopping
[ info     ] : k4a_device_stop_cameras(). k4a_device_stop_cameras stopped

wes-b, 2019-05-28:
Is this related to the device #375 is filed against? I am not able to repro this. After waking the PC, I do need to clear errors in the viewer, stop the camera, but starting again works as expected.

MsDerekMa, 2019-05-29:
Yes, this might be related to #375, it is on that same PC. The other Linux PC I have I need to make sure to close the device completely before trying to start the stream again, but I am able to start streaming again without On my other PC I am seeing that I need to close the device, but am then able to start streaming again without reconnecting the device.

wes-b, 2019-05-30:
I am not able to reproduce on my PC. Going to treat this as a one off until we get more data.

closed 2019-05-30 by wes-b