microsoft/Azure-Kinect-Sensor-SDK #375: On Linux the device doesn't enumerate if the power is connected second

MsDerekMa, 2019-05-24:
**Describe the bug**
If the power is connected second, a Linux system won't recognize the attached device.

**To Reproduce**
1. Connect a USB-A to C cable to the computer and plug it into the device.
2. Connect the power cable to the device.
3. Use the AzureKinectFirmwareTool to list the device.
4. The tool says no devices are connected.

**Expected behavior**
The order of the USB vs Power shouldn't matter and the device will enumerate if both cables are connected.

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04.2 LTS
 - SDK Version:  1.1.0-beta.0
 - Firmware version: 1.6.987014

MsDerekMa, 2019-05-24:
This appears to only affect one of my computers. The other computer everything works as expected. One other side effect is that instead of giving an error about not being able to open the device when running without the power connection, it just doesn't show up. In addition after the Firmware Update step where the device is rebooted, the device never enumerates.

MsDerekMa, 2019-05-28:
This is a computer with an Asus P8Z77-V PRO/THUNDERBOLT motherboard. The USB ports are from the Intel Z77 Chipset and are USB 3.1 Gen 1. The ASMedia host controller does not appear to work at all.

MsDerekMa, 2019-05-28:
USB View when the Kinect is working:
[image]

USB View when the Kinect is not working:
[image]

It looks like the USB hub in the device is enumerating, but that the devices downstream of that are not.

wes-b, 2019-05-30:
I can't repro this. Lets treat this as a one off issue until we get a signal this is happening on more than 1 PC.

closed 2019-05-30 by wes-b

lianghongzhuo, 2021-04-22:
I also have this problem. When Azure Kinect died (it died quite ofen), I have to unplug the power and USB, and first plug in the power adapter, second plug in the USB to the pc. The other way around will lead to unable to start the camera.
The USB cable is a custom 3-meter usb-c to USB-A cable. (I'm using ubuntu 18.04 with ros)

michael-createc, 2021-11-26:
Over 2 years later and this exact issue is still very much present on Jetson Nanos running Ubuntu 18.04