microsoft/Azure-Kinect-Sensor-SDK #384: k4aviewer fails to open microphone

Brent-A, 2019-05-28:
**Describe the bug**
When I try to open a device using k4aviewer, I get an error "Failed to start microphone: unable to open device!"

**To Reproduce**
On my home PC,
Open Azure Kinect Viewer (either 1.0.2 or 1.1.0-beta.0)
Open the device
Click Start
I get the pop-up dialog with "Failed to start microphone: unable to open device!"

**Expected behavior**
No error message, and streaming mic data.

**Logs**
[k4a log.txt](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/3229405/k4a.log.txt)

**Screenshots**
[image]

[image]

[image]

**Desktop (please complete the following information):**
 - OS with Version:  Windows 1903 
 - SDK Version:  e.g. 1.1.0-beta.0
 - Firmware version:  
```
[ info     ] : K4A SDK version:     1.1.0-beta.0
[ info     ] : Serial Number:       000070792012
[ info     ] : RGB Sensor Version:  1.6.98
[ info     ] : Depth Sensor Version:1.6.70
[ info     ] : Mic Array Version:   1.6.14
[ info     ] : Sensor Config:       6109.7
[ info     ] : Build type:          Release
[ info     ] : Signature type:      MSFT
```

**Additional context**

joylital, 2019-05-28:
@Brent-A might be container ID mix up. Try following steps:
- Right clicking on "Azure Kinect Microphone Array", and select "Uninstall device".  (it is the parent device you have selected in above window)
- Detach and reattach the sensor.
- Restart Azure Kinect Viewer and try again

Brent-A, 2019-05-28:
You can see from my screenshot that it looks to be in the same container as the rest of the Azure Kinect components.

xthexder, 2019-05-28:
This seems to reproduce on both of my devices as well with the 1.1.0-beta.0 viewer.

tesych, 2019-08-06:
The issue should be resolved by Windows update.

closed 2019-08-06 by tesych

Josephcs88, 2019-08-14:
I have updated Windows to 18362.267 and SDK v1.1.1
[image]

But, the same failed message is still there. The following figure was from running Azure Kinect Viewer v1.1.1
[image]

When I rebuilt the k4aviewer by gitting clone the source code at GitHub. There is no output from those sub-windows of Infrared, Depth, Color camera, and IMU as follows.
[image]
Did I miss something? 
Your help will be very appreciated.

zhuoyuan00, 2023-05-18:
> The issue should be resolved by Windows update.

The same problem with ubuntu20.04. What should I do?

eric-tc, 2023-11-20:
Hi @zhuoyuan00 ,
I have the same problem on ubuntu 20.04 did you find a solution?
Thanks