microsoft/Azure-Kinect-Sensor-SDK #387: k4aViewer failure: "Failed to initialize audio backend: incompatible device!"

wes-b, 2019-05-30:
**Describe the bug**
On Linux, open k4aviewer with not device connected. It should open without error. Instead the user gets a popup that says ""Failed to initialize audio backend: incompatible device!" Funny thing is that the user has not yet selected a device. So the "incompatible device" should not yet be known.

**To Reproduce**

opalmu, 2019-08-16:
I met with the same issue. Could you double check the cable that you used to connect the power and USB 3.0 port? I mistakenly used the USB 3.0 cable to connect the power. And that mistake generates me the same error as yours.

wes-b, 2019-08-19:
Our expectation is that this is only a Linux issue.

@tesych What is the issue tracking the Windows 10 bug around Kinect Audio not working after an update?

wes-b, 2019-08-19:
Found it: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/384

ZhangAustin, 2020-05-31:
> Our expectation is that this is only a Linux issue.
> 
> @tesych What is the issue tracking the Windows 10 bug around Kinect Audio not working after an update?

same issue here on Ubuntu 19.10, libk4a1.4-dev.
no issues on windows 10.

ZhangAustin, 2020-05-31:
> Our expectation is that this is only a Linux issue.
> 
> @tesych What is the issue tracking the Windows 10 bug around Kinect Audio not working after an update?

same issue here on Ubuntu 19.10 with libk4a1.4-dev
works fine on Windows 10 though.