microsoft/Azure-Kinect-Sensor-SDK #388: Mac support

tim-depthkit, 2019-05-30:
I'd like to write cross platform software that makes use of the Azure Kinect. Currently macOS is not supported by this SDK, which leaves a large user base out of being able to use this sensor.

What is involved in supporting macOS?

wes-b, 2019-05-30:
Thanks @tim-depthkit for the feedback. Can you please head over to https://aka.ms/azurekinectfeedback to register your request? Please also capture the architecture that you would need.

closed 2019-05-30 by wes-b

omidb, 2021-02-12:
hi, any update on this?

D1no, 2021-11-07:
Any update on this? Also the link provided is dead.

This is the most upvoted user request for a "development kit" because developers happen to use the linux BSD based mac operating system.
[image]

rajkundu, 2021-12-14:
+1 - and could an updated link be posted? I'd like to show my support for this. Especially with Apple Silicon now – it would be great to run this on macOS.

PimTournaye, 2022-04-11:
+1, not sure why MS is staying silent on this? It's still the highest rated request on the feedback forum

rajkundu, 2022-07-26:
By the way, the updated link is https://feedback.azure.com/d365community/idea/33a28bf8-f624-ec11-b6e6-000d3a4f0da0
Please vote!

rajkundu, 2023-01-24:
@wes-b @Brent-A Is there any possible way to get a response from anyone at Microsoft on this issue? It is currently the single most requested feature on the feedback forum, with 262 votes.

For my own purposes, a high-power, small-form-factor host computer like the Mac Mini would simply be perfect. Its hardware is now more than capable of handling the Azure Kinect DK, and the sole reason that it cannot do so is due to Microsoft's (unexplained) refusal to release a macOS version of the Depth Engine. If I knew how to fix this issue, then believe me - I would spend as long as needed to do so. However, I neither know whether nor how to accomplish this, especially given that the Depth Engine is proprietary.

Given the wide range and capability of Apple's latest computers, I imagine that many others - at least 261 others, to be exact - feel the same way as me. I understand Microsoft's decision to initially decline macOS support in 2019, but a generational leap in computing power has occurred since then. Over the last three years, Apple released the M1, M1 Pro, M1 Max, M1 Ultra, M2, M2 Pro, and M2 Max processors, all of which eclipse the best processors from 2019.

Furthermore, many higher education and research institutions have high-volume contracts with Apple, so the importance of supporting Apple computers cannot be understated. Support for macOS and Apple Silicon would substantially increase the ease, reach, and usefulness of the Azure Kinect DK.

_Please_ consider revisiting this issue in light of the significant computing advancements which have been made over the past three years. Any response at all from Microsoft would be _incredibly_ appreciated by not only me, but by so many others as well.

mattbeghin, 2023-11-19:
As a software editor, we'd like to integrate depth cameras. There are not much on the market. Intel RealSense stopped producing part of their line. Having Kinect Azure as a cross-platform solution would be very interesting for our user community.

djpiper28, 2024-04-17:
If I work on a Mac port, would Microsoft being willing to merge it?