microsoft/Azure-Kinect-Sensor-SDK #394: Corrupt color frames on subordinates

DavidMRCS, 2019-06-06:
**Describe the bug**
When I do a capture with 3 Kinects, each on a different PC, I find that the first frame on the subordinates is corrupt/all black. A second bad frame on subordinates comes around 7/8 frames later (varies per capture).

**To Reproduce**
I used my own modified capture software.

Do a capture using these settings
k4a_device_configuration_t deviceConfig = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
deviceConfig.color_format = K4A_IMAGE_FORMAT_COLOR_BGRA32;
deviceConfig.color_resolution = K4A_COLOR_RESOLUTION_1080P;
deviceConfig.depth_mode = K4A_IMAGE_FORMAT_DEPTH16;
deviceConfig.camera_fps = K4A_FRAMES_PER_SECOND_5;
deviceConfig.synchronized_images_only = true;
deviceConfig.wired_sync_mode = (k4a_wired_sync_mode_t)SyncMode;     // from command line
if (SyncMode == K4A_WIRED_SYNC_MODE_SUBORDINATE)
           deviceConfig.subordinate_delay_off_master_usec = uDelay; // from command line

Capture using a master and 2 subordinates across 3 PC's.

Extract color images from .bin files.

**Expected behavior**
I'd expect all images to be valid.

**Logs**

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - Windows 10 1903
 - SDK Version:  1.0.1
 - Firmware version:  
Current Firmware Versions:
  RGB camera firmware:      1.6.98
  Depth camera firmware:    1.6.70
  Depth config file:        5006.27
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

**Additional context**

wes-b, 2019-06-07:
@DavidMRCS I cannot reproduce this problem on newer hardware. My repro attempt was on a single machine. You might consider attempting my repro process to see if you can still reproduce the issue. You can view the mkv files with K4aViewer.exe to look at each frame.

```
start "sub 1" cmd /c bin\k4arecorder.exe --device 1 -l 20 --imu off -r 5 -e -6 -d WFOV_UNBINNED -c 1080p --external-sync Subordinate sub1.mkv ^> sub1.mkv.txt
ping 127.0.0.1 -n 2 ^> null
start "sub 2" cmd /c bin\k4arecorder.exe --device 2 -l 20 --imu off -r 5 -e -6 -d WFOV_UNBINNED -c 1080p --external-sync Subordinate sub2.mkv ^> sub2.mkv.txt
ping 127.0.0.1 -n 3 ^> null
start "master" cmd /c bin\k4arecorder.exe --device 0 -l 20 --imu off -r 5 -e -6 -d WFOV_UNBINNED -c 1080p --external-sync Master master.mkv ^> master.mkv.txt
```

I suspect your preproduction devices with ```Depth config file: 5006.27``` are the ones reporting bad frames. You want devices that are ```Depth config file: 6109.7``` or better.
```
Current Firmware Versions:
  RGB camera firmware:      1.6.98
  Depth camera firmware:    1.6.70
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft
```

wes-b, 2019-06-13:
New devices were delivered.

closed 2019-06-13 by wes-b