microsoft/Azure-Kinect-Sensor-SDK #396: Document external capture stablization

wes-b, 2019-06-06:
**Describe the bug**
Multi device capture with manual exposure can take 8 capture to stabilize.

**Additional context**
Auto exposure in multi device is strongly discouraged because time synchronization will always be fluctuating due to the moving exposure time.

tesych, 2019-06-27:
This is documented in [Azure Kinect documentation](https://docs.microsoft.com/azure/Kinect-dk/troubleshooting#multi-device-synchronization)

wes-b, 2019-06-27:
We might want to move topic from the above location to just after here: https://docs.microsoft.com/en-us/azure/Kinect-dk/retrieve-images#configure-and-start-the-device. As stabilization requirements seem to align better with getting samples for the first time rather than trouble shooting.

wes-b, 2019-06-27:
Feel free to close if you disagree

tesych, 2019-07-09:
updated the documentation

closed 2019-07-09 by tesych