microsoft/Azure-Kinect-Sensor-SDK #397: Document supported usb controllers

wes-b, 2019-06-06:
**Describe the bug**
The only USB controllers that are known with work with the Azure Kinect are Intel, Texas Instruments, and some Renesas (720201 was only chipset tested). Others were tested but did not work.

vpenades, 2019-06-10:
Additionally, it could be nice if the SDK includes the equivalent of the "Kinect Configuration Verifier" tool that was originally included with the KinectV2 SDK.

That tool included a list of valid/invalid USB controllers as well as other hardware incompatibilities (like certain GPUs). The list was supposed to be updateable online, but it seems it was never maintained beyond the initial release. It could be nice if this time we can have a list in json or xml.

tesych, 2019-06-26:
the instructions were added to the documentation at [Azure Kinect Documentation](https://docs.microsoft.com/en-us/azure/Kinect-dk/troubleshooting#usb3-host-controller-compatibility)

closed 2019-06-26 by tesych