microsoft/Azure-Kinect-Sensor-SDK #404: Support recording K4A_IMAGE_FORMAT_COLOR_BGRA32 image type

yijiew, 2019-06-11:
**Is your feature request related to a problem? Please describe.**
BGRA color image type is convenient for computer vision algorithms. It is very likely for a customer to config the device/playback to provide the K4A_IMAGE_FORMAT_COLOR_BGRA32 type. However, the record API doesn't support recording this type. 

**Describe the solution you'd like**
Record API supports recording K4A_IMAGE_FORMAT_COLOR_BGRA32 image type. 

**Describe alternatives you've considered**
Or provide a convenient API for the user to convert the K4A_IMAGE_FORMAT_COLOR_BGRA32 to other recording-supported types like: K4A_IMAGE_FORMAT_COLOR_NV12, K4A_IMAGE_FORMAT_COLOR_YUY2...

**Additional context**
None

RoseFlunder, 2019-10-17:
The playback API allows the user to specify an image format for color image conversion:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/include/k4arecord/playback.h#L375

Maybe this issue can be closed.

yijiew, 2019-10-17:
@RoseFlunder : You are right. This API allows the user to specify the format during playback. But imagine a scenario that you try to run the sensor in the live mode. You need to directly consume the color image in RGBA format. You set the mode to stream the RGBA format live. But meanwhile, you also want to write the live image to mkv so that you can repro some bugs in the future by playing back the clip. Unfortunately, the recording API doesn't allow to store the RGBA format image.

wes-b, 2019-12-06:
We should consider
-  allowing BGRA32 to be recorded and then played back
- allowing MJPEG to be recorded and played back as BGRA32

xthexder, 2019-12-06:
> We should consider
> 
> * allowing BGRA32 to be recorded and then played back
> * allowing MJPEG to be recorded and played back as BGRA32

Recording MJPEG and playing back as BGRA32 is already supported.

Recording BGRA32 is extremely impractical because of the file size. It's double the size of YUY2, which puts it at 248MB/s for 1080p@30fps, and exceeds the read/write speed of the majority of (if not all) spinning disks.

Currently the only thing stopping BGRA32 from being recorded is this switch statement: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/src/record/sdk/record.cpp#L85
It can be recorded today using custom tracks as well.
If we had some way to record the images before they're decompressed to BGRA, that would be ideal for recording while processing live data.

yijiew, 2019-12-06:
The reason I created this bug is because of the use case I mentioned above: we need to use BGRA data meanwhile recording the data to file. What @xthexder pointed out is a very valid point. The file size will be too big if we store the data in BGRA32 format. There are two possible ways to mitigate the problem: 
1. Like what @xthexder suggested, have a way to record the data before they are decompressed to BGRA. 
Or 2. Enable the record API to convert the BGRA back to MJPEG or other supported format to record (just like what we did for the playback API). We need to run an additional BGRA to MJPEG conversion this way. But since recording is running on a separate thread anyway, this additional operation might be acceptable.

wes-b, 2019-12-10:
We discussed this offline. Instead of decompressing from native MJPEG to BGRA32 in the SDK and then recompressing to MJPEGin the recorder, we will instead refer users to stream in MJPEG, record in MJPEG, and provide sample code to convert MJPEG to BGRA32. We will also enable BGRA32 recording for users with PC's and file systems that can handle that demand.

PR #967 (2019-12-11): Allow recording BGRA color, and add warning for slow disk speeds

## Fixes #404 

### Description of the changes:
- Allow BGRA format captures in recording
- If the write queue is more than 2 seconds behind, print a warning about slow disk speed.

### Before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [x] I [built my changes](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) locally
- [x] I ran the [unit tests](https://github.com/Microsoft/Azure-Kinect-Sensor [... cut]
closed 2019-12-11 by xthexder