microsoft/Azure-Kinect-Sensor-SDK #405: xorg-dev linux build dependency missing in docs

37, 2019-06-11:
Quick / simple one. 

When following the instructions to build from source on linux I found xorg-dev is a dependency not listed alongside the others. 

See: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md#linux-dependencies

My system information: 
Ubuntu 18.04 running i3 with x. 

Command entered:
Step2. 

`    cmake .. -GNinja
`

Errors: 

```
CMake Error at extern/glfw/src/CMakeLists.txt:222 (message):
  The RandR library and headers were not found
-- Configuring incomplete, errors occurred!
```

Attempted installing just libxrandr-dev (sudo apt install libxrandr-dev), which progressed the install to another error: 

```
-- Could NOT find Vulkan (missing: VULKAN_LIBRARY VULKAN_INCLUDE_DIR)
-- Using X11 for window creation
CMake Error at extern/glfw/src/CMakeLists.txt:231 (message):
  The Xinerama library and headers were not found
```

So I just installed the full xorg-dev:
sudo apt install xorg-dev

and the SDK built successfuly. 

```
-- Configuring done
-- Generating done
-- Build files have been written to: /home/noom/Code/kinect/Azure-Kinect-Sensor-SDK/build
```

## Suggested action: 
add xorg-dev to list of build deps.

MsDerekMa, 2019-06-13:
The "building.md" documentation references a list of dependencies and references a script that will install all dependencies.  The list of dependencies in the documentation is what is out of date, the script should install everything appropriately. We should consider just referencing the script itself as the source of truth for the list of dependencies.

schultetwin1, 2019-06-14:
Thank you for reporting! I'm on it

schultetwin1, 2019-06-14:
I've change the documentation to just point towards the installation script. That script is run in CI and so should be kept up to date.

PR #414 (2019-07-10): Updates documented linux dependencies
## Fixes #405 

### Description of the changes:
- Changes building.md to point Linux dependencies to bootstrap-ubuntu.sh

### Before submitting a Pull Request:
- [X] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [ ] I [built my changes](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) locally
- [ ] I ran the [unit tests](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/testing.md)
- [ ] I ran the [functional tests](https://github.com [... cut]
closed 2019-07-10 by Brent-A