microsoft/Azure-Kinect-Sensor-SDK #406: Include instructions for linux binaries and k4a-tools in SDKv1.1

37, 2019-06-11:
After building the SDK for linux it becomes apparent you need the linux binaries / depthengine.so.1.0 file for the SDK to work properly, as per [/docs/depthengine.md](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/depthengine.md). 

For example, running /build/bin/viewer_opengl: 

```
Started opening K4A device...
[2019-06-11 15:12:18.898] [error] [t=17739] ../src/dynlib/dynlib_linux.c (86): dynlib_create(). Failed to load shared object libdepthengine.so.1.0 with error: libdepthengine.so.1.0: cannot open shared object file: No such file or directory
...
[2019-06-11 15:12:18.908] [error] [t=17732] ../src/sdk/k4a.c (741): depth_start(device->depth, config) returned failure in k4a_device_start_cameras()
Failed to start cameras!
Press [enter] to exit.
```

According to the docs the binaries are not yet available for Linux, though should be around for SDKv1.1, which I've just built. 

I did some digging and found [#290](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/290) discussing binary naming and versioning conventions and the mention of binaries being available for linux at http://packages.microsoft.com/

> This is all done. Debian packages are available on packages.microsoft.com. Instructions on how to add packages.microsoft.com are [here](https://docs.microsoft.com/en-us/windows-server/administration/linux-package-repository-for-microsoft-software)
> 
> I'm working on getting them updated to include more recent versions of Ubuntu

Looking further into it I managed to get the binaries set up on my system (Ubuntu 18.04). 

## Setting up binaries on Linux: 

For setup instructions for other distros see https://docs.microsoft.com/en-us/windows-server/administration/linux-package-repository-for-microsoft-software.

#### Ubuntu 18.04 instructions

```
wget https://packages.microsoft.com/config/ubuntu/18.04/packages-microsoft-prod.deb
sudo dpkg -i packages-microsoft-prod.deb
sudo apt-get update
sudo apt-get install k4a-tools
```

Success! I could now run the SDK examples. 

[image]
[image]

## Suggested actions:
Add instructions for setting up linux repositories and installing the k4a-tools .debs to /docs/depthengine.md

wes-b, 2019-06-13:
#293

wes-b, 2019-06-13:
#7 Is the original issue about this.

schultetwin1, 2019-06-14:
@37 thank you for the awesome write-up and nice investigative skills. I have added instructions for how to install debian packages to docs/usage.md.

Here is the PR for that change  #412

PR #412 (2019-06-14): Adds linux installation docs to usage.md
## Fixes #406 

### Description of the changes:
- Adds documentation for installing Linux Debian packages

### Before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [ ] I [built my changes](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) locally
- [ ] I ran the [unit tests](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/testing.md)
- [ ] I ran the [functional tests](https://github.com/Microsoft/A [... cut]
closed 2019-06-14 by 37