microsoft/Azure-Kinect-Sensor-SDK #416: k4a.hpp cannot access the k4a_capture_t handle from the capture class

yijiew, 2019-06-18:
**Is your feature request related to a problem? Please describe.**
The C++ wrapper for capture is handy and easy to use. I would like to use the C++ wrapper in body tracking. However, there is no way I can access the k4a_capture_t handle from the class. Since the recording API requires the raw k4a_capture_t handle, there is no way I can use the capture class with the k4a recording feature. 

**Describe the solution you'd like**
Create a "handle()" function like the image class that expose allow users to access the k4a_capture_t handle from the capture class. 

**Describe alternatives you've considered**
None.

**Additional context**
None.

UnaNancyOwen, 2019-07-29:
This issue also occurs when using Azure Kinect Body Tracking SDK with C ++ wrapper of Azure Kinect Sensor SDK.
The second argument of <code>k4abt_tracker_enqueue_capture()</code> requests <code>k4a_capture_t</code> handle. But, there is no way to reference <code>k4a_capture_t</code> handle from <code>k4a::capture</code> class.
We need to add a way to access <code>k4a_capture_t</code> handle to <code>k4a::capture</code> class.
```patch
+   k4a_capture_t get_handle() const noexcept
+   {
+       return m_handle;
+   }

private:
    k4a_capture_t m_handle;
};
```

RoseFlunder, 2019-08-06:
@UnaNancyOwen
If this method would be added, would I need to call the release capture method for each k4a_capture_t returned by this method or is it enough when only the wrapper class releases the original capture once with its reset method?

UnaNancyOwen, 2019-08-06:
@RoseFlunder I’m not sure. I think it is maybe enough to just call <code>k4a::capture::reset()</code>.

skalldri, 2019-08-09:
@yijiew : I'm going to take this bug. This is blocking some work on the Azure Kinect ROS Driver (adding Body Tracking support) and I'd like to just quickly add a capture::handle() function (like the image:: class has) to provide access to the underlying k4a_capture_t handle.

PR #626 (2019-08-10): Add a capture::handle() function to k4a.hpp

## Fixes #416

### Description of the changes:
- Adds a `capture::handle()` method to the C++ wrapper, allowing access to the underlying k4a_capture_t when using the C++ wrapper. This is needed to allow the C++ wrapper and the Body Tracking SDK to work together

### Before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [x] I [built my changes](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) locally
- [x] I ran the [ [... cut]
closed 2019-08-10 by skalldri