microsoft/Azure-Kinect-Sensor-SDK #419: OpenGL error: 'gl_FragColor' undeclared when showing point cloud in k4aviewer

xthexder, 2019-06-18:
**Describe the bug**
On a fresh installation of Ubuntu 18.04 with the default graphics drivers, switching to the point cloud view in K4AViewer results in a 'gl_FragColor' undefined error.
The current usage is deprecated and should be declared as an `out` variable in the shader.

**To Reproduce**
1. Install k4a debian packages on a fresh Ubuntu 18.04 installation.
2. Open a sample recording (I used OFFICE_SAMPLE_WFOV_UNBINNED.mkv)
3. Switch to the 3D / point cloud view.

**Expected behavior**
The point cloud view should be displayed.

**Logs**

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04 with latest Nouveau driver
 - SDK Version:  1.1.0
 - Firmware version:  N/A, recording file was used.

tesych, 2019-07-01:
@williampmsft can you please provide more information.

xthexder, 2019-07-01:
I should be able to take on this bug this week. I think we have all the info we need to fix this.

closed 2019-07-15 by xthexder

opalmu, 2019-08-20:
Hello @xthexder @tesych. I have met with this issue again on a fresh installed Ubuntu 18.04. The error message is the same as the one shown above. 

**Describe the bug**
On k4aviewer, switching to 3D mode generates the following error message.

**To Reproduce**
Call k4aviewer and start the device, click on 3D mode and the error shows up
```
terminate called after throwing an instance of 'std::logic_error'
  what():  Shader compilation error: 
0:15(2): error: `gl_FragColor' undeclared
0:15(2): error: value of type vec4 cannot be assigned to variable of type error
```

**Expected behavior**
From what I have seen when I called k4aviewer in Windows 10, there is a 3D point cloud scene of the environment. 

**Logs**
```
terminate called after throwing an instance of 'std::logic_error'
  what():  Shader compilation error: 
0:15(2): error: `gl_FragColor' undeclared
0:15(2): error: value of type vec4 cannot be assigned to variable of type error
```

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04
 - SDK Version:  v1.1.1 (I guess it should be v1.1.1 since I directly clone the github repo on Aug. 20, 2019)
 - Firmware version: N/A

The problem is that I also installed Kinect Azure SDK in Windows 10 with the same computer and the 3D mode works all good there. Do you know how could I solve this issue on Ubuntu system?

xthexder, 2019-08-20:
The fix for this was not merged until after the 1.1.1 release, and should be included in the latest 1.2.0 beta build.

You're most likely seeing this because you're using the default Nouveau drivers that come with Ubuntu.
If you install the vendor specific graphics drivers (such as Intel or Nvidia, depending on your machine), the viewer should run without issue.

opalmu, 2019-08-20:
@xthexder Thanks for your response! I checked that I installed Intel driver i915 in this Ubuntu. Also I have installed Mesa19.0.2 and OpenGL version 3.0. After rebooting the computer the issue still exists, would you please point me any other directions to go? Many thanks! 
Here I attach the screen shot of my terminal.
[image]

xthexder, 2019-08-20:
The depth engine for Azure Kinect requires OpenGL 4.4, which isn't fully supported by Mesa.
Mesa and Nouveau are similar in that they're open-source implementations of the graphics driver.

You'll likely need to uninstall the mesa driver in order for the proprietary Intel driver to run if you have both installed.
Unfortunately I don't have a lot more info on how to install the correct drivers, but there's plenty of other resources online.

PR #505 (2019-07-15): Various K4A Viewer fixes

## Fixes #419 #424 #453 #504

### Description of the changes:
- Fix IMU graph label being delayed #504 
- Fix point cloud fragment shader deprecated usage #419
- Fix playback of PASSIVE_IR recordings #424 
- Add better error checking for failed window creation #453 

### Before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [x] I [built my changes](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) locally
- [ ] I ran the [unit tests](ht [... cut]