microsoft/Azure-Kinect-Sensor-SDK #422: Race condition on multiple threads calling logger_register_message_callback

Brent-A, 2019-06-19:
**Describe the bug**
If two threads simultaneously call `logger_register_message_callback`, there is a possible race where one thread will allocate g_user_logger_cb_info, and another thread will attempt to access that global before it is allocated.

Thread 1 
* Calls `logger_register_message_callback`
* increments `g_user_logger_cb_info_ref` (`count_of_registered_callbacks` is 1)
* Thread gets context switched away before completing `g_user_logger_cb_info = new …`

Thread 2
* Calls `logger_register_message_callback`
* increments `g_user_logger_cb_info_ref` (`count_of_registered_callbacks` is 2)
* Crashes dereferencing NULL during `g_user_logger_cb_info->callback`

**To Reproduce**
Get very unlucky in timing when starting a multi-threaded process that attempts to initialize logging on multiple threads at the same time.

**Expected behavior**
App doesn't crash 😊

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: Windows 1903
 - SDK Version:  1.1.0
 - Firmware version:  N/A

**Additional context**

wes-b, 2019-06-19:
@Brent-A definitely a bug, but why is logger_register_message_callback being called twice from different threads? Did you find this through code inspection? I would like to better understand your usage pattern. The function will fail if you try to use different callback functions.

Brent-A, 2019-06-19:
I'm wrapping K4A in a class that gets activated by an external framework on multiple threads. As the class author, I have no control over how often, and on what threads, I get activated.

wes-b, 2019-06-24:
Fix here is to spin when multiple registrations are happening at the same time to ensure initialization completes.

PR #475 (2019-07-11): Added global init and rwlock

## Fixes #422

### Description of the changes:
- Adds a new reader/writer lock object
- Adds a global one time initialization helper
- Updates the logger to use the lock and init routines

 

### Before submitting a Pull Request:
- [X] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [X] I [built my changes](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) locally
- [X] I ran the [unit tests](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/tes [... cut]
closed 2019-07-11 by Brent-A