microsoft/Azure-Kinect-Sensor-SDK #427: USB Extenders / Extensions

wes-b, 2019-06-20:
USB Extenders are a nice tool for connecting multiple devices to 1 PC or to extend the distance between the host PC and the Azure Kinect.

While USB technology is implemented everywhere and generally easy to use, it can also come with its own quarks when trying to connect a lot of devices, or trying to use a cable length that is longer than the USB Specification intended it to be used for. In the case of the Azure Kinect it internally has USB 2.0 and USB 3.0 hubs. The 7 MIC array is attached to the USB 2.0 hub while both the color camera and the depth camera are connected to the USB 3.0 hub. To complicate matters USB 2.0 has response time requirements that make extenders harder to use as they have to find ways to minimize the response delays while trying to add greater cable lengths. As a result it is difficult to extend the cable length of a USB device.

The plan is to update this topic as more information about extenders becomes available.
 
### Failure Class 1 – Container ID changes ###
One class of failures is that some USB Extenders implement a USB hub within their cable. When the hub firmware reports the hub port is _not user connectable_, it allows the Windows USB stack to treat children connected to it as hard wired to the hub and implies the port cannot be detached from the hub. When this happens Windows assigns the hubs Container ID to all children because it thinks they are 1 monolithic device. The Windows platform needs these container ID’s to match the container ID’s imbedded in the device descriptor so that when a depth camera is opened, the color camera on the same device can also be opened.

This can be confirmed with the ```USBView``` tool (downloadable from Microsoft) where the hub port has the property is ``` Is Port User Connectable: no```.

The error pattern generated from Azure Kinect logs would look like this if you are hitting this problem
```
[2019-06-20 14:43:46.938] [error] [t=37768] ..\..\src\color\mfcamerareader.cpp (###): RuntimeClassInitialize(). Failed to find color camera in open camera: 0xc00d36d5
[2019-06-20 14:43:46.940] [error] [t=37768] ..\..\src\color\color.cpp (##): SUCCEEDED( Microsoft::WRL::MakeAndInitialize<CMFCameraReader>(&color->m_spCameraReader, (GUID *)container_id)) returned failure in color_create()
[2019-06-20 14:43:46.940] [error] [t=37768] ..\..\src\sdk\k4a.c (###): color_create( device->tick_handle, container_id, serial_number, color_capture_ready, handle, &device->color) returned failure in k4a_device_open()
```

To work around this issue, the SDK can be recompiled to use the first color camera it finds regardless of the container ID. The drawback to this solution is that the SDK will only support 1 Azure Kinect at a time. Look at the function FindEdenColorCamera in mfcamerareader.cpp.

closed 2019-06-20 by wes-b

tim-depthkit, 2019-06-27:
I'd like to use active USB extenders like the Icron Spectra 3001-15 or Icron Spectra 3022. It appears that these extenders do not preserve the container ID as you describe, which is evident by several missing devices when looking at the Azure Kinect in the connection ID view in Device Manager.

I've not yet tried the suggested work-around to modify the SDK, but I'll post here with test results after doing so.

If it works, it would be nice if the SDK allowed the user to choose which method is used to connect to the USB device: either using the first color camera found, or by container ID. This would allow for the most flexibility in deployments of the device where it is not feasible to have the host computer close to the sensor.

valentinptc, 2019-07-10:
Have you found some updates for making active USB extension cables work for multiple cameras? Or have you found an active extension cable that does not implement an USB hub?

wes-b, 2019-07-11:
I have not, but I am hoping we hear about more from the community. Our testing is limited to what I can borrow and far from complete.

valentinptc, 2019-07-11:
We tested a couple of cables today and found a 10 Meters/32.8 Feet extension cable that worked:
https://www.amazon.com/gp/product/B00DMFFL2W/

I hope that helps you guys as well.

tylergibson, 2019-08-20:
@valentinptc I am using these cables as well, they work great for data.  But I can't seem to get the Kinect to use them for power.  I use the original AC adapter with a USB A to 5.5mm barrel adapter to supply input power to the active cable, and have a USB power meter connected to output - Confirmed polarity is correct and current draw is correct.

But when I connect it to the Kinect using the USB A to C cable, it only draws the base amperage and I get the flashing amber light (even though there is PLENTY of power available to it).  Is this a handshake issue for power delivery?  Any resolution here?

wes-b, 2019-08-20:
The spec for USB A to C is not sufficient to power the Kinect alone and the reason for the supplemental power via the barrel connector. To power without a barrel connector you must use USB C to C.

RoseFlunder, 2019-08-20:
We are also searching for some longer (at least 4m) USB C to C cables to connect multiple kinects to a single PC. My university will order one of these for testing:
[https://www.amazon.de/Tether-Tools-TetherPro-USB-Datenkabel-gerader/dp/B079WWSR5G/](https://www.amazon.de/Tether-Tools-TetherPro-USB-Datenkabel-gerader/dp/B079WWSR5G/)

I could confirm its working after it gets delivered.

taeinkwon, 2019-08-21:
Hi, Do you know any compatible USB A-C cables for Kinect? Maybe more than 5 m? I tested with "[Baseus 3A Artistic Striped Series (A - C, 5m)" cable but my computer can not recognize the device when I use this cable. I also use the original power supply cable with this one. Are their any constraints for the cable?

UnaNancyOwen, 2019-08-28:
By the way, I want a spare cable for supply sub-power.
Do you know product specification of this connector?

wes-b, 2019-09-09:
> By the way, I want a spare cable for supply sub-power.
> Do you know product specification of this connector?

#745

alan-newnex, 2019-10-10:
Check this A to C active cable out, https://www.ntcdistributing.com/usb-3-1/active-extension-cable/firenex-ulink-c-active-cable-a-m-to-c-m-8-m-and-16-m/
It's tested to extend Azure Kinect up to 16m.
https://www.newnex.com/technology-azure-kinect-extension.php

tylergibson, 2019-10-12:
@alan-newnex - Does this cable also carry power, or just data?  In my testing no A-C cable can carry enough amperage, only C-C can delivery the required power.

I am currently using https://www.amazon.com/Cable-Matters-Active-Extension-Female/dp/B00DMFFL2W/ref=sr_1_3?keywords=cable+matters+usb+extender+powered&qid=1570839947&sr=8-3

These from Cable Matters, and using an AC extension cord to the 5vdc power supply with a DC splitter to power the extension cable (using yet another plug adapter) and the Kinect (with a plug adapter from Aliexpress).  Not pretty at all, but very inexpensive (total cost of about 60 dollars per camera) gets me a bit over 10m of distance per camera.

yuemasy, 2019-12-03:
> @alan-newnex - Does this cable also carry power, or just data?  In my testing no A-C cable can carry enough amperage, only C-C can delivery the required power.
> I am currently using [...]

@tylergibson, thanks for providing this solution. it's very attractive. I would like to double check that  Kinect will be connected to the cable's type A female, and the cable need an extra power supply.  Kinect will use its original plug adapter to supply power for itself.  So I need to buy the cable and another power adapter, right? Thank you very much.

tylergibson, 2019-12-03:
@yuemasy yep correct

ibrachahine, 2019-12-17:
My team and I have bought a USBC(M)-USBA(M) passive cable of 4.6 meters and a USBA(F)-USBA(M) extender and the combination works perfectly even for 30fps.

 USBC(M)-USBA(M) used:
https://www.bol.com/nl/p/tether-tools-cuc3215-org-usb-kabel-4-6-m-3-2-gen-1-usb-a-usb-c-oranje/9200000086904005/?bltgh=m89teua-f2dL9aDNvhBNOA.1_4.6.ProductTitle

USBA(F)-USBA(M) used: 
https://www.bol.com/nl/p/tether-tools-cu3017-usb-kabel-5-m-3-2-gen-1-usb-a-oranje/9200000039912899/?bltgh=m89teua-f2dL9aDNvhBNOA.1_4.10.ProductTitle

In europe, it is harder to find reliable cables of high quality compared to the US. You have to be lucky.

N.B. Microsoft does not yet support extenders for Kinect.

reneschulte, 2019-12-20:
What is the Kinect team using? No need for an approved of official cable recommendation, just want to know what works and what does not.

alan-newnex, 2019-12-20:
> @alan-newnex - Does this cable also carry power, or just data? In my testing no A-C cable can carry enough amperage, only C-C can delivery the required power.
>
> I am currently using [...]

The Newnex cable https://www.amazon.com/dp/B0113Q0V4E is bus powered, so there is no need for any external power source to drive the repeater circuit.

alan-newnex, 2019-12-20:
Newnex has also released an active cable that support USB Power Delievery for Kinect Azure. https://www.ntcdistributing.com/usb-repeater/active-cable/usb-c-to-usb-c-active-cable-for-azure-kinect-with-pd-10m/
With this cable, a user only needs one cable from a USB-C host to the Kinect, it doesn't need extra USB power input, which is extremely desired when a user is trying to extend the Kinect.

tylergibson, 2019-12-22:
@alan-newnex that's awesome!  The price is unfortunately a bit prohibitive for me as a hobbyist, but looks like a super solid commercial solution :)

BennoMouma, 2020-02-20:
@ibrachahine I'm also looking for a solution for this from the netherlands. Did you try to connect multiple camera's and did that work? And you did you into any other problems in the meantime?

metervara, 2020-09-09:
We used these cables on a project with two kinects:
https://www.bhphotovideo.com/c/product/1545201-REG/vaddio_440_1005_051_usb_3_0_active_optical.html

They're expensive but work great. These do not power the kinects though, so we still use the dedicated power supply.

SaraHackett13, 2020-09-29:
Does this mean that USB 3.0 hubs cannot be used to support multiple Azure Kinect devices? I plan to use a USB 3.0 hub to connect each camera to a single PC. But I will power each camera with the power cable/adapter that comes with the Kinect. Is this feasible?

tim-depthkit, 2020-09-30:
You may run into USB bandwidth issues if running multiple Azure Kinects from a single USB3.0 hub, as the hub will need to share bandwidth amongst all connected devices. Ideally each Azure Kinect would have a dedicated full speed USB 3.0 connection (5 Gbps). We've had success with the [this PCIe](https://www.startech.com/en-us/cards-adapters/pexusb3s44v) card that provides 4 full speed ports by utilizing two controllers.

Chris45215, 2020-11-16:
Cable length has always been a challenge when I have used the devices - though sometimes it's the power cable's length that is the bigger issue. We find that the sensors work best when mounted above head height. But the power cable is only 5 feet long - so we have always needed power extension cables. We don't want to put strain on the power cable or connection within the sensor, so we usually end up with something taped to a wall or tripod. That's far from ideal.

If Microsoft releases a subsequent version of the sensors, a longer power cable would be very welcome. Also, a power brick that doesn't cover multiple outlets would be a major improvement too. The current power brick covers 2 outlets by itself, and the cable sticking out further means that it actually covers 3 outlets along a power strip. Please at least have the cable stick straight out (like Apple arranges it with all their chargers), rather than up or downwards, so the unit only takes the space of 2 plugs.

It may seem like a small issue, but it becomes a significant hindrance when setting up numerous sensors.

gdosu, 2020-12-02:
> To work around this issue, the SDK can be recompiled to use the first color camera it finds regardless of the container ID.

May I take some advice on this part? I haven't tried modifying DLL files, so any little tips would be appreciate.

greatwallet, 2021-03-08:
> We are also searching for some longer (at least 4m) USB C to C cables to connect multiple kinects to a single PC. My university will order one of these for testing: [...]

Have you gotten any results?

greatwallet, 2021-03-08:
> Newnex has also released a 10m (33 ft.) active cable that supports USB Power Delivery for Kinect Azure. https://www.ntcdistributing.com/usb-repeater/active-cable/usb-c-to-usb-c-active-cable-for-azure-kinect-with-pd-10m/
> With this cable, a user only needs one cable from a USB-C host to the Kinect, it doesn't need extra USB power input, which is extremely desired when a user is trying to extend the Kinect.

I don't have any USB-C port on my host, so does an usb-A to usb-C converter work in this case?

alan-newnex, 2021-03-08:
> I don't have any USB-C port on my host, so does an usb-A to usb-C converter work in this case?

Unfortunately no, only a native USB-C port will be able to support PD feature. It will keep the data transfer, but you will still need to connect the USB power adapter for the power.

noorfatma, 2021-09-13:
Will this extender work for extending the kinect.  The newnex is taking time to be shipped to India. Is there any additional driver required for these ones.

https://www.amazon.in/Ugreen-Superspeed-Active-Extension-Repeater/dp/B01FQ88CHS/?_encoding=UTF8&pd_rd_w=KFefo&pf_rd_p=47cc7285-f327-4110-bcbf-06d303677587&pf_rd_r=H3VR9TKS235T6N4TK2FA&pd_rd_r=1f059fd7-b426-4f20-8543-c59682830a73&pd_rd_wg=IvUQ4&ref_=pd_gw_ci_mcx_mr_hp_d