microsoft/Azure-Kinect-Sensor-SDK #429: Setting depth mode to WFOV_Unbinned in CSharp example fails

rolandsmeenk, 2019-06-21:
**Describe the bug**
Setting DepthMode to WFOV_Unbinned returns K4A_RESULT_FAILED

**To Reproduce**
1. Open Microsoft.AzureKinect.Examples.WPFViewer
2. Change the DeviceConfiguration initialization to use
 DepthMode = DepthMode.WFOV_Unbinned
3. Running the application results in a K4A_RESULT_FAILED

**Expected behavior**
The Kinects starts providing the depth data as requested.

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 10.0.17134 Build 17134
 - feature/CSharp branch
 - Firmware version:  
Current Firmware Versions:
  RGB camera firmware:      1.5.92
  Depth camera firmware:    1.5.66
  Depth config file:        6109.7
  Audio firmware:           1.5.14

**Additional context**
The Kinect was tested to work with WFOV_Unbinned when running the Azure Kinect Viewer v1.1.0

Brent-A, 2019-07-11:
Hi @rolandsmeenk. We just merged the CSharp branch to develop so it's getting a bit more stable.

If you can get logs from the SDK we should be able to figure out more about what's going on. 
You can enable [logging to a file](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/develop/group___logging.html) using some environment variables.

Also make sure that the program has the following DLLs with it:
* Microsoft.Azure.Kinect.Sensor.dll
* k4a.dll
* depthengine_1_0.dll

MsDerekMa, 2019-07-12:
Thanks, I was just able to try this out.  It looks like it is failing because the default FPS is 30 FPS. 30 FPS is not supported with WFOV Unbinned.  You can check our [supported operating modes].(https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification#depth-camera-supported-operating-modes)

However, this pointed out that we need to do some work on our exceptions and logging to try to make it more clear where the errors are coming from. I created Issues #488 and #489 to enable better diagnostics. Please add to those or create a new issue if you need more diagnostics that aren't covered.

closed 2019-07-12 by MsDerekMa

Sacristan, 2020-02-20:
I had the very same error - used on system that has Intel internal GPU. Circumstances...

```
using (Tracker tracker = Tracker.Create(deviceCalibration, new TrackerConfiguration() { ProcessingMode = TrackerProcessingMode.Gpu, SensorOrientation = SensorOrientation.Default }))
```

```
TrackerProcessingMode.Gpu -> TrackerProcessingMode.Cpu
```