microsoft/Azure-Kinect-Sensor-SDK #430: Implement precision time sync by reading PTS/USB frame correlation from firmware

wes-b, 2019-06-24:
This is for scenarios when a solution needs high precision timestamps that allow the user to know exactly when an image was captured as referenced to the system time, rather than the device time.

**Is your feature request related to a problem? Please describe.**
For late stage rendering, precision between the system time and the device time is needed. This is needed so that system time can measure overall system performance and latency.

#198 captures the first part of addressing this, while this issue specifically address fixing the precision part of the issue.

catid, 2019-10-21:
Maybe worth looking into this: `WinUsb_StartTrackingForTimeSync`

wes-b, 2019-10-23:
> Maybe worth looking into this: WinUsb_StartTrackingForTimeSync

Yes, this is great for windows, support on Linux is the question. I think we will need USB driver enhancements on Linux to do this there.