microsoft/Azure-Kinect-Sensor-SDK #436: Document how to avoid depth camera interfience when using multiple camera's

wes-b, 2019-06-24:
**Describe the bug**
When using multiple depth camera's in synchronized captures, depth camera captures should be offset from one another by 160us or more.

tesych, 2019-08-22:
it is documented as part of the [Hardware specification](https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification#external-synchronization)

closed 2019-08-22 by tesych

iammarvelous, 2019-10-30:
Hi, I was going through documents about multiple devices sync. I finally ended up here because I was wondering how to set sync-delay. I think [this document](https://docs.microsoft.com/en-us/azure/Kinect-dk/record-external-synchronized-units) should also be updated to avoid confusion. To be clear, I want to check with you that the following is the correct way to do multiple devices sync:
```
# for master
k4arecorder.exe --device 0 --external-sync master output-1.mkv
# for subordinate-1
k4arecorder.exe --device 1 --external-sync subordinate --sync-delay 0.16 output-2.mkv
# for subordinate-2
k4arecorder.exe --device 2 --external-sync subordinate --sync-delay 0.32 output-3.mkv
```

wes-b, 2019-10-30:
> Hi, I was going through documents about multiple devices sync. I finally ended up here because I was wondering how to set sync-delay. I think this document should also be updated to avoid confusion. [...]

Hi @iammarvelous can you open a new issue please?

iammarvelous, 2019-10-30:
@wes-b Done. See ##862