microsoft/Azure-Kinect-Sensor-SDK #441: Getting the floor plane

vpenades, 2019-06-27:
**Is your feature request related to a problem? Please describe.**

I've looked throught the code of Kinect Azure DK and body tracking SDK and I've noticed there's no way to retrieve the floor plane anywhere.

**Describe the solution you'd like**

Typically, floor plane has been available along with tracked skeleton bodies, but I consider it as a core feature, not something that belongs to an extended SDK, so to me, I woud like this feature to be available in the core SDK.

**Describe alternatives you've considered**

If not in the core SDK, at least it should be available in the body tracking SDK...

**Additional context**

Floor plane is important to transform any detected feature from camera space to world space.

It's typically used on tracked skeleton bodies, But it's not the only use case.

Even for plain image manipulation, being able to transform points from depth camera space to world space allows many interesting artistic effects.

cdedmonds, 2019-07-19:
This feature request for body tracking has been moved to user voice. 

https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/38188435-ground-plane-support

closed 2019-07-19 by cdedmonds

Thaina, 2023-01-05:
@cdedmonds Did UserVoice site is already closed ?