microsoft/Azure-Kinect-Sensor-SDK #443: K4A body tracking SDK doesn't track hands and doesn't provide hand status data

rfilkov, 2019-06-27:
**Is your feature request related to a problem? Please describe.**

The body tracking in Kinect SDK 2.0 tracks the wrists, hands, index fingers and thumbs. The new K4A body tracking SDK tracks the arms down to the wrists, as far as I see.

Apart of that, the previous BT SDK provides the hand state (open, closed or even lasso) of each hand. The current one does not provide this data, yet.
 
**Describe the solution you'd like**

Please include at least the hands into the tracked body joints list (even Kinect SDK 1.x tracked hands), as well as the hand state (open or closed) for each hand. This would provide good means for interaction with virtual objects. Even inferior body tracking SDKs provide this data.

**Describe alternatives you've considered**

The alternative would be to use custom or 3rd party solutions for hand and finger tracking.

A better alternative would be to get hand and finger tracking added later to the K4A body tracking SDK.

**Additional context**

I'm not sure this is the correct place to report body tracking issues. If not, please direct me to the right place, or forward this feature request to the right people.

rfilkov, 2019-07-15:
Any update on this one?

cdedmonds, 2019-07-16:
This is in consideration for a future release. For feature requests please use UserVoice. I have started a new issue here that you can upvote. 

https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/38154784-hands-support-in-body-tracking

closed 2019-07-16 by cdedmonds

Thaina, 2024-02-08:
> This is in consideration for a future release. For feature requests please use UserVoice. I have started a new issue here that you can upvote.
> 
> https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/38154784-hands-support-in-body-tracking

Uservoice already closed