microsoft/Azure-Kinect-Sensor-SDK #454: Build on clean Ubuntu 18.04 succeeds;device enumerates; however cannot `start`.

eric-schleicher, 2019-07-02:
**Describe the bug**

Example applications can enumerate, but not start kinect device after successful setup &  build on fresh installation of Ubuntu 18.04 (x64)

Appears to be permission related on the USB port  possible this is just an Udev rule?

**To Reproduce**
Steps to reproduce the behavior:
1. On vanilla Ubuntu 18.04 LTS (desktop) Host
2. Clone and setup build environment per official [instructions page](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md)
   used the `scripts/bootstrap_ubuntu.sh` shell file to prepare the environment
3. Build examples as instructed on ^ instructions page
4. Run (from `build/bin`) `./k4aviewer`;  note that no usb devices are enumerate and console show permission warning
  ```
rosuser@vega:~/Downloads/Azure-Kinect-Sensor-SDK/build/bin$  ./k4aviewer 
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/002/008: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
[2019-07-02 14:57:52.405] [error] [t=12861] ../src/usbcommand/usbcommand.c (305): find_libusb_device(). Unable to open LIBUSB at index 0 
[2019-07-02 14:57:52.405] [error] [t=12861] ../src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-07-02 14:57:52.405] [error] [t=12861] ../src/sdk/k4a.c (129): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()

  ```

#### Run with elevated permissions
1. In a console enter: `sudo ./k4aviewer`
2. Select the device from available dropdown list, Click 'open device'
6. Select green 'Start' button
7. :(
[image]

**Expected behavior**
Expecting device to open device and stream!

**Screenshots**

when attempting to run  sample applications; they are not able to open the device, following dmesg output occurs

[image]

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04 LTS 
  `Linux vega 4.19.8-041908-generic #201812080831 SMP Sat Dec 8 13:34:18 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`
 - SDK Version:  
   git commit `7153c3c0a66219a884feeb1c18c1d04b61cc6493`
 - Firmware version:  
  [image]

**Additional context**

wes-b, 2019-07-03:
This does sound like a udev issue. But our expectation is that ```su ./k4aviewer``` is enough for the viewer to run successfully.

Also make sure you are seeing a white LED on the back of the device to ensure you are properly connected to USB 3.0 and power.

We also have a sample udev, see https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#linux-device-setup

eric-schleicher, 2019-07-03:
Yes, I get the LED as expected.  Once you close the error dialog, I do see that audio data is flowing (graph), but no IMU or camera(s)

I'll take look at your udev file.

wes-b, 2019-07-03:
Something else to consider is if the PC USB Host controller is compatable. We know there are issues with Container ID's on Windows and certain USB 3.X controllers. We also know that some just fail after a few seconds of operation.

If you come up with a good way to list host controllers on Linux it would be great to share as the topic below is Windows focused.
https://docs.microsoft.com/en-us/azure/Kinect-dk/troubleshooting#usb3-host-controller-compatibility

tesych, 2019-07-30:
There's no repro for this issue. Please provide more information and we will revisit this.

closed 2019-07-30 by tesych

martinakos, 2019-10-03:
I have the same problem. Ubuntu 18.04 I have copied the 99-k4a.rules to /etc/udev/rules.d/ and I do get a solid white light in the back of the device. Have you found a solution for this problem?

wes-b, 2019-10-03:
Failing to start is usually caused by the depth engine but being installed

martinakos, 2019-10-03:
yes, if I disable the depth camera in the GUI I can see the color camera. do I need to install anything else besides k4a-tools?

RoseFlunder, 2019-10-03:
You need the azure kinect sdk and the installation is described here:
https://docs.microsoft.com/en-us/azure/kinect-dk/sensor-sdk-download

martinakos, 2019-10-04:
I did follow that installation guide. I had also installed sudo apt install libk4a1.1-dev 
yet I got the "failed to start device!" error.

wes-b, 2019-10-04:
Please share your logs and open a new issue. Are you trying to call start on a machine you have remoted to?

seva100, 2019-10-09:
@martinakos I have had exactly the same problem on Ubuntu 18.04, and updating graphics driver helped. I used the standard system dialogue (Software & Updates -> Additional Drivers) and updated from nvidia-418 to nvidia-430, then made a reboot. Afterwards, depth camera in k4aviewer started working.

murez, 2020-07-29:
I tested on Ubuntu 20.04 with the bionic deb packages, and it works well by the command `sudo k4aviewer`.

jacquesj1, 2020-11-18:
Hello,
[image]

I have had exactly the same problem on Ubuntu 18.04, but the rgb cam don't work.
My graphic cards is RTX2080ti (drivers 455).
The log:

[2020-11-18 23:39:51.405] [error] [t=3038] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    684400 type:Depth
[2020-11-18 23:39:51.439] [error] [t=3038] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    717733 type:Depth
[2020-11-18 23:39:51.472] [error] [t=3038] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    751066 type:Depth
[2020-11-18 23:39:51.505] [error] [t=3038] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    784400 type:Depth
[2020-11-18 23:39:51.539] [error] [t=3038] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    817733 type:Depth
[2020-11-18 23:39:51.567] [error] [t=3042] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (269): capturesync_get_capture(device->capturesync, capture_handle, timeout_in_ms) returned failure in k4a_device_get_capture()
[image]