microsoft/Azure-Kinect-Sensor-SDK #457: Linux install fail because it cant find ' libk4a1.1-dev' package.

valdivj, 2019-07-03:
I cant complete my install on LINUX because when I run  : sudo apt install libk4a1.1-dev.
I get the error :cant find package?
Any ideas?

wes-b, 2019-07-03:
Hi @valdivj without any of the supplemental information our issue template asks for I am poking around in the dark. Have you pointed your PC at the Microsoft Package Repository? 

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#debian-package

valdivj, 2019-07-03:
Im setting up the Kinect for Azure on Ubunta 18.04.
I followed the instructions to set up the Microsoft Package Repository.
And this works: sudo apt install k4a-tools
But this : sudo apt install libk4a1.1-dev returns  :cant find package?
Do I not have my Microsoft Package Repository setup correctly?

schultetwin1, 2019-07-03:
Can you try `apt search k4a` and share the output of the command? I'm curious what you see? This is what I see

```
matt@maschu-tower:~$ apt search k4a
Sorting... Done
Full Text Search... Done
k4a-tools/bionic,now 1.1.0-release amd64 [installed]
  Tools for Azure Kinect Development

libk4a1.1/bionic,now 1.1.0-release amd64 [installed,automatic]
  Dynamic Libraries for Azure Kinect Runtime

libk4a1.1-dev/bionic,now 1.1.0-release amd64 [installed]
  Headers and cmake files needed for Azure Kinect Development
```

valdivj, 2019-07-04:
This is what it returns:
Sorting... Done
Full Text Search... Done
k4a-tools/bionic,now 1.1.1 amd64 [installed]
  Tools for Azure Kinect Development

libk4a1.1/bionic,now 1.1.1 amd64 [installed,automatic]
  Dynamic Libraries for Azure Kinect Runtime

libk4a1.1-dev</font>/bionic 1.1.1 amd64
  Headers and cmake files needed for Azure Kinect Development

schultetwin1, 2019-07-05:
I'm not sure why apt search would be able to find libk4a1.1-dev but apt install would then say "can't find package".

Let's try two things:

1. Try copy and pasting "libk4a1.1-dev" from the search results into the `sudo apt install` command. Maybe there is something weird going on with the hyphen in the command.

2. Can you share the full `sudo apt install` command you ran and all the output?

schultetwin1, 2019-07-05:
I also just noticed that in [our documentation](https://docs.microsoft.com/en-us/azure/Kinect-dk/sensor-sdk-download#linux-installation-instructions) we say to install

`sudo apt install libk4a1.1-dev`.

That period at the end should not be part of the package name. Please only run the command

`sudo apt install libk4a1.1-dev`

I will change the documentation to show this as well.

schultetwin1, 2019-07-05:
MicrosoftDocs/azure-docs#34590 has this change to remove the period from the install command.

valdivj, 2019-07-05:
It works now.
Curse that period.
Apt search k4a now returns:

Sorting... Done
Full Text Search... Done
k4a-tools/bionic,now 1.1.1 amd64 [installed]
  Tools for Azure Kinect Development

libk4a1.1/bionic,now 1.1.1 amd64 [installed,automatic]
  Dynamic Libraries for Azure Kinect Runtime

libk4a1.1-dev/bionic,now 1.1.1 amd64 [installed]
  Headers and cmake files needed for Azure Kinect Development

valdivj, 2019-07-05:
Curse that period.
apt search k4a now returns this:

Sorting... Done
Full Text Search... Done
k4a-tools/bionic,now 1.1.1 amd64 [installed]
  Tools for Azure Kinect Development

libk4a1.1/bionic,now 1.1.1 amd64 [installed,automatic]
  Dynamic Libraries for Azure Kinect Runtime

libk4a1.1-dev/bionic,now 1.1.1 amd64 [installed]
  Headers and cmake files needed for Azure Kinect Development

Thank You

schultetwin1, 2019-07-05:
Awesome! Glad to hear it's working. We've updated our docs to remove that period. That should be live within the week.

closed 2019-07-05 by schultetwin1

valdivj, 2019-07-05:
Hey I don't know if this is a coincident but the Azure for Kinect stopped working after I got 'libk4a1.1-dev' to install. It wont recognize there is a Kinect for Azure at all now. I was able to at least run the viewer before but nothing now. I reinstalled and rebooted my Ubunta machine couple time but no luck. I even went and plugged in to my windows10 PC and it wont  recognize it either. It ran fine on the windows 10 machine before. Just a heads up. I will contact Microsoft for some tech help.

schultetwin1, 2019-07-08:
Hi valdivj, that sounds like a different issue. It should not be related to the version of the libk4a1.1-dev package you have installed. Please contact our hardware support team for help with this issue. You can find info about device support [here](https://docs.microsoft.com/en-us/azure/kinect-dk/support#azure-kinect-dk-device).

For some quick steps you can try on your own I would.

1. Make sure you are still plugged into a USB 3.0 port.
2. Try hard powering off and on your computer (to fully clear any weird USB state on your machine).
3. Check the output of dmesg when you plug in your device.

wes-b, 2019-07-08:
Also make sure the power LED on back is solid white. Also be sure to use the supplied power supply.

valdivj, 2019-07-08:
I have contacted Microsoft support. The LED on the back is blinking slowly. What does that mean?

wes-b, 2019-07-08:
https://docs.microsoft.com/en-us/azure/kinect-dk/hardware-specification#indicators