microsoft/Azure-Kinect-Sensor-SDK #459: Kinect Device Crash and Unresponsive

thedevguy, 2019-07-03:
Using two new Kinect DK devices. Getting the same result with both. Connected them to my PC into two separate USB ports. I am able to reproduce this issue every single time. When I run Azure Kinect Viewer, both cameras are available. After I open either device and simply attempt to start it, I get IMU errors. Below is the Log report:

[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:1233733333
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:1233766666
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:1233800000
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:1233833333
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:1233866666
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:1233900000
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:1233933333
[ warning  ] : depth_engine_thread(). Dropping depth image due to bad timestamp at startup
[ warning  ] : imu_capture_ready(). IMU startup dropped last 368 samples, the timestamp is too large
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     16088 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     49422 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     82755 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    116088 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    149422 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    182755 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    216088 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    249422 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    282755 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    316088 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    349422 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    382755 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    416088 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    449422 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    482755 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    516088 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    549422 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    582755 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    616088 type:Depth
[ warning  ] : queue_disable(). Waiting for blocking call to complete
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : capturesync_add_capture(). Capture Error Detected, Depth 
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ error    ] : usb_cmd_io(). Error calling libusb_bulk_transfer for initial tx, result:LIBUSB_ERROR_NO_DEVICE
[ error    ] : usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, NULL, 0, p_data, data_size, NULL, &cmd_status) returned failure in usb_cmd_write()
[ error    ] : usb_cmd_write(colormcu->usb_cmd, DEV_CMD_IMU_STREAM_STOP, NULL, 0, NULL, 0) returned failure in colormcu_imu_stop_streaming()

wes-b, 2019-07-05:
Hi @thedevguy, what OS are we talking about. The log indicates you are not getting color images from the camera. Please make sure you are not using an extra cables, make sure you are powered sufficiently with the device power supply provided in box. Then please review the USB host controller compatibility info below, and finally look for third party software and drivers that may be impacting your ability to use the color camera in the camera app (assuming we are talking about Windows here).

https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting#usb3-host-controller-compatibility

thedevguy, 2019-07-06:
OS: Windows 10
Not using any extra cables. Using what came in the box. Both devices are plugged into the wall charger using the wall charger that came in the box.

thedevguy, 2019-07-06:
I've purchased a new USB card from Amazon for my PC (https://www.amazon.com/gp/product/B00HJZEA2S) that has 4 discreet Renesas controllers. Once I get the card I'll report on the issue.

thedevguy, 2019-07-09:
I received and installed the usb controller that meets published specifications. The specifications are here: https://www.startech.com/Cards-Adapters/USB-3.0/Cards/PCI-Express-USB-3-Card-4-Dedicated-Channels-4-Port~PEXUSB3S44V

The kinect is detected however using the Kinect Viewer neither camera are available to open. When I try to refresh devices I get this error:

[ error    ] : find_libusb_device(). Unable to open LIBUSB at index 0 
[ error    ] : usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[ error    ] : depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
[ error    ] : find_libusb_device(). Unable to open LIBUSB at index 1 
[ error    ] : usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[ error    ] : depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()

[image]

thedevguy, 2019-07-09:
I have done the following trouble shooting steps:
1. installed new drivers for the usb card
2. unsinstalled the kinect dk sdk software and reinstalled it
3. scarified small goat

thedevguy, 2019-07-09:
Here is a screen capture of how the device manager looks and how the viewer doesnt show the device listed.

[image]

wes-b, 2019-07-09:
The yellow triangles next to "Azure Kinect Depth Camera" and "Azure Kinect 4k Camera" indicate the two cameras are not enumerated successfully. You should be able to double click on the errored out devices to determine what the error is. Also please try multiple USB ports on the card, don't assume they are all exactly the same.

thedevguy, 2019-07-14:
Wes here are what the devices show when I double click on the error-ed out devices. I have tried every port I have on the USB controller with no change to the error.

[image]
[image]
[image]
[image]

thedevguy, 2019-07-14:
Can the Kinect DK work with Windows PCs that have AMD CPUs?

wes-b, 2019-07-15:
These devices should have WinUsb installed on them. I can't see the full history via the images. Not sure if you have some 3rd party software messsing with the WinUsb driver or not. You might try updating the driver and specifically telling Windows to use WinUsb.inf.

wes-b, 2019-07-15:
And yes, AMD CPU's work fine.

lllbrucccce, 2019-07-19:
I had a similar problem,
[image]

[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  42100000
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  42133333
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  42166666
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  42200000
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  42233322
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  42266666
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  42300000
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  42333333
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : imu_capture_ready(). IMU startup dropped last 408 samples, the timestamp is too large
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     17733 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     51066 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     84400 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    117733 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    151066 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    184400 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    217733 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    251066 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    284400 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    317733 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    351066 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    384400 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    417733 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    451066 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    484400 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    517733 type:Depth
[... cut ...]
[ warning  ] : capturesync_add_capture(). Capture Error Detected, Depth 
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     17733 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     51066 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     84400 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    117733 type:Depth

wes-b, 2019-07-19:
Try this:
1) Open device manager - Windows Button + r -> devmgmt.msc
2) Unplug all Kinects
3) Set View-> Devices by Type
4) Set View -> Show Hidden Devices
5) Under "Cameras" right click on all instances of "Azure Kinect 4k Camera" and select "Uninstall Device"
6) Under "USB Serial Bus devices" right click and select "Uninstall Device" for all "Azure Kinect 4k Camera" and "Azure Kinect Depth Camera" instances present.
7) Plug the Kinect in and try again.

lllbrucccce, 2019-07-19:
> 试试这个：
> 
> 1. 打开设备管理器 -  Windows按钮+ r  - > devmgmt.msc
> 2. 拔掉所有Kinects
> 3. 按类型设置视图 - >设备
> 4. 设置视图 - >显示隐藏的设备
> 5. 在“Cameras”下右键单击“Azure Kinect 4k Camera”的所有实例，然后选择“卸载设备”
> 6. 在“USB串行总线设备”下右键单击并为所有“Azure Kinect 4k Camera”和“Azure Kinect Depth Camera”实例选择“卸载设备”。
> 7. 将Kinect插入并再试一次。

I tried. It didn't work.

lllbrucccce, 2019-07-19:
> Try this:
>
> 1. Open device manager - Windows Button + r -> devmgmt.msc
> 2. Unplug all Kinects
> 3. Set View-> Devices by Type
> 4. Set View -> Show Hidden Devices
> 5. Under "Cameras" right click on all [...]

I tried. It didn't work.

wes-b, 2019-07-22:
@lllbrucccce I didn't notice this before, but ```[ warning ] : CFrameContext(). No metadata attached to the sample``` is unique to your log and makes your issue different. Please lets start a new issue for this. Does the camera work fine with the camera app? I am wondering if you have some other camera software installed that may be filtering the camera images.

wes-b, 2019-07-22:
I am closing this issue as it seems like the original issue has been resolved. We can reopen if new information is available.

thedevguy, 2019-07-22:
This issue has not been resolved. Please keep open.

wes-b, 2019-07-22:
Sure, what is your current status after trying:
```
Try this:
Open device manager - Windows Button + r -> devmgmt.msc
Unplug all Kinects
Set View-> Devices by Type
Set View -> Show Hidden Devices
Under "Cameras" right click on all instances of "Azure Kinect 4k Camera" and select "Uninstall Device"
Under "USB Serial Bus devices" right click and select "Uninstall Device" for all "Azure Kinect 4k Camera" and "Azure Kinect Depth Camera" instances present.
Plug the Kinect in and try again.
```

tesych, 2019-07-30:
Closing the issue due to no new information.

thedevguy, 2019-08-01:
Please reopen. I havent had access to the PC with the issue until now.

thedevguy, 2019-08-01:
I tried the following and there isn't an option for "cameras". I've attached what I see in Device Manager.

[image]
[image]

thedevguy, 2019-08-01:
I realized I was looking at the wrong view, Resources by Type, rather than Devices by Type. I followed the steps provided and did not have any different results.

[image]

thedevguy, 2019-08-01:
When starting the viewer I dont see the device and see the following error:

[ error    ] : find_libusb_device(). Unable to open LIBUSB at index 0 
[ error    ] : usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[ error    ] : depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()

[image]

thedevguy, 2019-08-01:
@tesych would you please reopen this issue. Thank you.

thedevguy, 2019-08-01:
Using the Non-Renesas USB Controller the device is detected in the viewer however it crashes and I get these errors:

[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 56ms (this may be transient).
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  82400000
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  82433333
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  82466666
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  82500000
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  82533333
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  82566666
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  82600000
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  82633333
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  82666666
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  82700000
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  82733333
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  82766666
[ warning  ] : imu_capture_ready(). IMU startup dropped last 448 samples, the timestamp is too large
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     16088 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     49422 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     82755 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    116088 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    149422 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    182755 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    216088 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    249422 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    282755 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    316088 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    349422 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    382755 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    416088 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    449422 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    482755 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    516088 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    549422 type:Depth
[ warning  ] : queue_disable(). Waiting for blocking call to complete
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : capturesync_add_capture(). Capture Error Detected, Depth

ultrafro-ptc, 2019-08-05:
@thedevguy I was using Startech USB 3.0 card (PEXUSB3S44V) with Renesas drivers and had the same issue and here's what fixed it for me:

Unplug all kinects. Go into device manager, look for the Renesas USB root hubs, uninstall the devices and delete the drivers when prompted. Install the Renesas drivers as administrator. Re-plugin the kinect and verify that the 4k camera as well as the depth have no yellow flags. 

The kinect streams were then viewable in the tool. I tried the above steps without deleting the drivers and it didn't work. My guess is that the automatically installed renesas hub drivers were sticking around and not being replaced by the drivers startech provided. The hub name also changed to eXtensible afterwards.

This is a screenshot of my Renesas hubs, note that the hub says eXtensible now:
[image]

and the kinect in device manager when viewed by container:
[image]

Sorry if you've already tried this

wes-b, 2019-08-05:
@ultrafro-ptc thanks for the image. The ```(Microsoft)``` at the end of ```Renesas USB 3.0 eXtensible Host Controller -1.0 (Microsoft)``` also tells you that you are running the Microsoft driver rather than the Renesas driver.

thedevguy, 2019-08-05:
I’ll check it out thank you.
-Ali

closed 2019-08-23 by wes-b

rajk97, 2019-10-15:
I had a similar experience at the beginning. But that I came to know that my laptop's installed webcam was being called upon somehow when I try to run the cameras. I then uninstalled the laptop's camera. Then upon lowering the protection standards in my Kaspersky AntiVirus, I was able to run the cameras smoothly. Weird but worked for me.

Krisonrik, 2019-11-11:
> @thedevguy I was using Startech USB 3.0 card (PEXUSB3S44V) with Renesas drivers and had the same issue and here's what fixed it for me:
>
> Unplug all kinects. Go into device manager, look for the [...]

 I have tried this solution. the StarTech driver performed even worse than microsoft drivers. I had to uninstall them to at least to be able to see the camera in Azure viewer. Btw, i'm using a StarTech PCI card with 4 slots.

barakooda, 2019-11-21:
I`m using 
https://www.siig.com/4-port-superspeed-usb-3-0-pcie-card-quad-core-lp.html
Chipset: Renesas uPD720202

Cant see kinect devices when they connect.

wes-b, 2019-11-21:
@barakooda are you talking about Windows or Linux?

USB Card [JU-P40811-S1 ](https://www.asrock.com/MB/Intel/X299%20OC%20Formula/index.asp) works fine. Awesome! Thank you.

>works fine :
https://www.asrock.com/MB/Intel/X299%20OC%20Formula/index.asp

What do you mean by this? mainboard and USB  card works great? USB3 ports on the main board work great?

>kinect devices not turn on \ recognized when they connected.
>Z370-A PRO

What do you mean by this? USB ports don't work for you? Do you know what USB Host controllers are used here?

barakooda, 2019-11-22:
Sorry for not being clear, I updated my text.
thanks.

wes-b, 2019-11-22:
> there is hardware section we can share our test results ?
> should we open one ?

@barakooda, thanks for the idea, https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/936 has been opened.