microsoft/Azure-Kinect-Sensor-SDK #460: CI Builds failing due to failure to download NASM

schultetwin1, 2019-07-05:
**Describe the bug**
All CI builds are currently failing due to a failure to download the NASM assembler.

**Logs**
```
##[section]Starting: Download NASM
==============================================================================
Task         : PowerShell
Description  : Run a PowerShell script on Linux, macOS, or Windows
Version      : 2.151.1
Author       : Microsoft Corporation
Help         : https://docs.microsoft.com/azure/devops/pipelines/tasks/utility/powershell
==============================================================================
Generating script.
========================== Starting Command Output ===========================
##[command]"C:\windows\System32\WindowsPowerShell\v1.0\powershell.exe" -NoLogo -NoProfile -NonInteractive -ExecutionPolicy Unrestricted -Command ". 'D:\a\_temp\ef53841f-dbca-4f30-b393-411a9b3bd582.ps1'"
Invoke-WebRequest : Unable to connect to the remote server
At D:\a\_temp\ef53841f-dbca-4f30-b393-411a9b3bd582.ps1:28 char:5
+     Invoke-WebRequest -Uri $url -OutFile $path -UserAgent "NativeClie ...
+     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidOperation: (System.Net.HttpWebRequest:HttpWebRequest) [Invoke-WebRequest], WebExc 
   eption
    + FullyQualifiedErrorId : WebCmdletWebResponseException,Microsoft.PowerShell.Commands.InvokeWebRequestCommand
 
##[error]PowerShell exited with code '1'.
##[section]Finishing: Download NASM
```

schultetwin1, 2019-07-05:
This download works fine on my work machine. It maybe related to networking on Azure, or maybe NASM is blocking our CI machines?

schultetwin1, 2019-07-05:
Note, I tried to ping nasm.us from Azure DevOps.

```
Pinging nasm.us [198.137.202.136] with 32 bytes of data:
Request timed out.
Request timed out.
Request timed out.
Request timed out.

Ping statistics for 198.137.202.136:
    Packets: Sent = 4, Received = 0, Lost = 4 (100% loss),
```

Note, DNS resolution is working. Checking routing next.

schultetwin1, 2019-07-05:
Downloads of NASM are now working.

However, in order to prevent this from happening in the future, we need to remove the need to download NASM. There are two options for that:

1. Start building on a docker image.

2. Create a tool installer.

schultetwin1, 2019-07-05:
Re-opening this as the issue to track building docker images for this build.

wes-b, 2019-10-02:
@schultetwin1 we are using Docker now, can this be closed?

schultetwin1, 2019-10-02:
@wes-b We are using Docker for Linux now, but not Windows. I'll rename and label the issue to be specific.

schultetwin1, 2020-04-23:
We will not be able to create a Docker Image for Windows because it would need to contain Visual Studio (which we can not redistribute), Closing this issue.

closed 2020-04-23 by schultetwin1