microsoft/Azure-Kinect-Sensor-SDK #462: Device not detected - Windows 10

neuropil, 2019-07-05:
**Describe the bug**

Device is connected via USB 3.0 and LED on back is solid white. However, Azure Kinect Viewer fails to detect the device.

**Desktop (please complete the following information):**
 - OS with Version: Windows 1803
 - SDK Version:  1.1.0
 - Firmware version:  

**Additional context**

Azure Microphone is detected in the Device Manager - but not the camera

wes-b, 2019-07-05:
Sounds like USB2.0 devices are working but not 3.0. Are you using a USB extender?

Other topic to consider is USB host controllers: https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting#usb3-host-controller-compatibility

lllbrucccce, 2019-07-19:
Switch to a USB3.0 interface

wes-b, 2019-07-19:
I am guessing this is not longer an issue, closing because there has been no update.

closed 2019-07-19 by wes-b

ryanmjacobs, 2019-08-12:
I know it's a faux pas to comment on a closed issue... but I just wanted to confirm this issue.

My Kinect Azure fails to be recognized on a Windows 10 host on a USB 2.0 port. Switching to a USB 3.0 port works fine.