microsoft/Azure-Kinect-Sensor-SDK #464: sensor_calibration for remote processing of body tracking data

billpottle, 2019-07-07:
**Is your feature request related to a problem? Please describe.**

Due to the high minimum system requirements, (ie, GTX 1070), many of our potential customers won't be able to get joint position data on their current laptops. A potential fix is to have them record an mkv file from their device and then upload to an azure instance which could extract and send back the joint position data. 

In k4abt_tracker_create(&sensor_calibration, &tracker), I cannot find any way in the API to query a file to get the sensor_calibration data, and I cannot otherwise find any way to create a tracker.

**Describe the solution you'd like**

I would like a way to get the sensor_calibration data from the mkv file. There is the k4a_playback_get_record_configuration() function, but that returns type k4a_record_configuration_t, which is similar to but different than the needed type k4a_calibration_t. 

**Describe alternatives you've considered**

As a workaround, I was able to use the configuration code in the 'build your first body tracking app' with the same settings the file was recorded with to open the kinect locally and get sensor_calibration that way. This works to get joint data from files recorded previously,  however, this would not work for files recorded in the field with arbitrary settings on multiple devices. 

**Additional context**

Brekel, 2019-07-07:
The following function from playback.h gets you a k4a_calibration_t from an MKV file, and I've succesfully used it to track bodies from file:
k4a_result_t k4a_playback_get_calibration(k4a_playback_t playback_handle, k4a_calibration_t *calibration);

billpottle, 2019-07-07:
Thanks @Brekel. Not sure why I couldn't find that in the documentation before. I plugged it in my code and can confirm it works.

closed 2019-07-07 by billpottle

Brekel, 2019-07-07:
I'm known to wander header files before ever reading manuals so it's quite possible I found it there primarily :)

billpottle, 2019-07-07:
Yes, I searched for the function on Google and I don't think it's in the docs yet. Thanks again.