microsoft/Azure-Kinect-Sensor-SDK #465: Kinect DK SDK k4arecorder.exe Access Denied Error on Windows 10

mikewiegand, 2019-07-08:
I  just received my new Kinect DK a few days back. I have successfully installed the SDK on Ubuntu 1804, the viewer/recorder seem to work great so far. But, on W10, I cannot seem to get around the following error:

C:\temp>"C:\Program Files\Azure Kinect SDK v1.1.0\tools\k4arecorder.exe" -l 5 output.mkv
libusb: error [winusbx_open] could not open device \\.\USB#VID_045E&PID_097C#000041292112#{A5DCBF10-6530-11D2-901F-00C04FB951ED} (interface 0): [5] Access is denied.
[2019-07-04 08:39:44.111] [error] [t=11648] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (305): find_libusb_device(). Unable to open LIBUSB at index 0
[2019-07-04 08:39:44.114] [error] [t=11648] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth_mcu\depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-07-04 08:39:44.115] [error] [t=11648] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (129): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
Runtime error: k4a_device_open() failed
Runtime error: k4a_device_get_serialnum(device, serial_number_buffer, &serial_number_buffer_size) returned 1

On Ubuntu, running in su seems to remedy this, but for W10, I am running the CMD window in admin mode. I am unclear as to why this error is occuring. From additional research, seems USB incompatability could also be the issue. My dev preference is VS, but C in Linux is doable if this is the path I should persue. Maybe my issue is a simple fix, but if it actually is the USB, that's a problem.

Thanks

wes-b, 2019-07-08:
This error pattern does not match the errors I have seen resulting from USB Host controller issues. It seems more like something else is using the camera exclusively. 

Please make sure you are powered with the provided power supply and the back power LED is solid white. Also please make sure you are not using any extension cable and just the provided cables.

mikewiegand, 2019-07-08:
Thanks for the response. I have only used the provided cables / power supply. Solid white has always been fine. No extension cords. The odd thing is it works as described on a NUC717 w/Ubuntu 1804. The viewer, recorder, all seems to be fine. On w10, the viewer works great (I think), it's the recorder which breaks down. I do not seen anything odd running as this Alien is brand new; it also throws the exact same exception on Surface Pro as well as Alien.

wes-b, 2019-07-08:
Hmm, if you had the viewer open then that might be to app using the camera exclusively. Can you confirm that viewer works or not. You might just reboot too first if you have not tried that. 

The viewer and the recorder both use the same SDK code so if one works you should expect to the other to also work.

mikewiegand, 2019-07-08:
OK, not sure what happened, but both the viewer and k4arecorder.exe command
seem to work fine...
Not sure what I fat fingered. Thank you so much!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

closed 2019-07-08 by mikewiegand

wes-b, 2019-07-08:
Great!

jingweihaha, 2019-09-23:
> Hmm, if you had the viewer open then that might be to app using the camera exclusively. Can you confirm that viewer works or not. You might just reboot too first if you have not tried that.
>
> The [...]

After closing Viewer, the recorder starts working properly.

GabrielQuik, 2019-10-25:
Had the same issue turned out to be just a cable issue, make sure you plug in both your cables to the correct sources ;)