microsoft/Azure-Kinect-Sensor-SDK #468: Cannot Add Microsoft.Azure.Kinect.Sensor 1.1.1 Nuget to UWP App

mikewiegand, 2019-07-09:
Following instructions from: https://docs.microsoft.com/en-us/azure/kinect-dk/build-first-app:

I created a vanilla UWP App in VS 2017 and tried to add the Microsoft.Azure.Kinect.Sensor Nuget Package and received the following error. 

NU1202: Package Microsoft.Azure.Kinect.Sensor 1.1.1 is not compatible with uap10.0.17763 (UAP,Version=v10.0.17763). 
Package Microsoft.Azure.Kinect.Sensor 1.1.1 does not support any target frameworks.

Seems no matter what values I enter for Target Version/Min Version as well as the sensor 1.0 versus 1.1.

Must this be a UWP app?

wes-b, 2019-07-09:
We don't currentlhy support UWP. Try creating a console app project.

Head over to https://aka.ms/azurekinectfeedback to vote for UWP support if you need it.

mikewiegand, 2019-07-09:
When I create a new C# console app, I get 

Could not install package 'Microsoft.Azure.Kinect.Sensor 1.1.1'. You are trying to install this package into a project that targets '.NETFramework,Version=v4.6.1', but the package does not contain any assembly references or content files that are compatible with that framework. For more information, contact the package author.				

Thank u

tesych, 2019-07-10:
@mikewiegand you would need to create new C++ console app since C# is not available yet.

mikewiegand, 2019-07-10:
That was it. Thank you!

closed 2019-07-19 by rabbitdaxi