microsoft/Azure-Kinect-Sensor-SDK #470: Kinect showing in Device Manager, but not Kinect Viewer

monkeykat, 2019-07-09:
I am having an issue where the Kinect camera is showing up in the Device Manager, but the Kinect Viewer just says "No available devices." The Kinect Viewer works fine on another computer, so is not defective.

Windows 10 1809
USB Host Controller: Intel USB 3.0 eXtensible Host Controller - 1.0

wes-b, 2019-07-09:
If you plug into a USB 2 port then *most* everything will show up in device manager. You want to make sure that you can see "Azure Kinect Depth Camera" and that none of them have a yellow triangle.  Make sure the power LED on the back of the device is solid white.

The Device manager view below is View->Device By Container.

[image]

monkeykat, 2019-07-09:
Thanks for getting back so quickly! I checked my Device Manager, and everything looks to be the same as yours. No yellow triangles and the depth camera is showing up. The LED on the back is solid white and the USB port is 3.0.

[image]

xthexder, 2019-07-10:
If you open the log in the viewer (Settings -> Show log dock), and set the log level to trace, do you get any output when hitting `Refresh Devices`?

monkeykat, 2019-07-11:
Yes I do. Here it is:

[ trace    ] : logger_t_create(). Created   logger_t 0000016364AAA4B0
[ trace    ] : k4a_device_t_create(). Created   k4a_device_t 0000016364B14A10
[ trace    ] : depthmcu_t_create(). Created   depthmcu_t 0000016364AF6190
[ trace    ] : usbcmd_t_create(). Created   usbcmd_t 0000016364AEC7C0
[ info     ] : find_libusb_device(). Container ID found: {8376c5f0-06bd-40c7-749f-dd00637c0a61} 
[ info     ] : populate_serialnumber(). Serial Number found 000207592012
[ trace    ] : usb_cmd_io(). XFR: Cmd=00000115, PayloadSize=255
[ trace    ] : colormcu_t_create(). Created   colormcu_t 0000016364AAA0F0
[ trace    ] : usbcmd_t_create(). Created   usbcmd_t 0000016364AECF30
[ info     ] : find_libusb_device(). Container ID found: {8376c5f0-06bd-40c7-749f-dd00637c0a61} 
[ info     ] : populate_serialnumber(). Serial Number found 000207592012
[ trace    ] : calibration_t_create(). Created   calibration_t 0000016364A9BC70
[ trace    ] : usb_cmd_io(). XFR: Cmd=00000111, PayloadSize=10240
[ trace    ] : capturesync_t_create(). Created   capturesync_t 0000016364AB57E0
[ trace    ] : queue_t_create(). Created   queue_t 000001636F16A560
[ trace    ] : queue_t_create(). Created   queue_t 000001636F169E80
[ trace    ] : queue_t_create(). Created   queue_t 000001636F16A2E0
[ trace    ] : depth_t_create(). Created   depth_t 0000016364A4D150
[ trace    ] : usb_cmd_io(). XFR: Cmd=00000201, PayloadSize=18
[ trace    ] : usb_cmd_io(). XFR: Cmd=00000201, PayloadSize=18
[ info     ] : ******************** Device Info ********************
[ info     ] : K4A SDK version:     1.1.0
[ trace    ] : usb_cmd_io(). XFR: Cmd=00000115, PayloadSize=255
[ info     ] : Serial Number:       000207592012
[ info     ] : RGB Sensor Version:  1.6.98
[ info     ] : Depth Sensor Version:1.6.70
[ info     ] : Mic Array Version:   1.6.14
[ info     ] : Sensor Config:       6109.7
[ info     ] : Build type:          Release
[ info     ] : Signature type:      MSFT
[ info     ] : ****************************************************
[ trace    ] : dewrapper_t_create(). Created   dewrapper_t 0000016364B007E0
[ trace    ] : queue_t_create(). Created   queue_t 000001636F169F70
[ trace    ] : usb_cmd_io(). XFR: Cmd=000000f2, PayloadSize=0
[ trace    ] : usb_cmd_io(). XFR: Cmd=0000000a, PayloadSize=0
[ trace    ] : color_t_create(). Created   color_t 0000016364AF3EC0
[ error    ] : RuntimeClassInitialize(). Failed to activate source in open camera: 0x80070005
[ error    ] : SUCCEEDED( Microsoft::WRL::MakeAndInitialize<CMFCameraReader>(&color->m_spCameraReader, (GUID *)container_id)) returned failure in color_create()
[ trace    ] : color_t_destroy(). Destroyed color_t 0000016364AF3EC0
[ error    ] : color_create( device->tick_handle, container_id, serial_number, color_capture_ready, handle, &device->color) returned failure in k4a_device_open()
[ trace    ] : queue_t_destroy(). Destroyed queue_t 000001636F169F70
[ trace    ] : dewrapper_t_destroy(). Destroyed dewrapper_t 0000016364B007E0
[ trace    ] : depth_t_destroy(). Destroyed depth_t 0000016364A4D150
[ trace    ] : queue_t_destroy(). Destroyed queue_t 000001636F16A560
[ trace    ] : queue_t_destroy(). Destroyed queue_t 000001636F169E80
[ trace    ] : queue_t_destroy(). Destroyed queue_t 000001636F16A2E0
[ trace    ] : capturesync_t_destroy(). Destroyed capturesync_t 0000016364AB57E0
[ trace    ] : calibration_t_destroy(). Destroyed calibration_t 0000016364A9BC70
[ trace    ] : usbcmd_t_destroy(). Destroyed usbcmd_t 0000016364AEC7C0
[ trace    ] : depthmcu_t_destroy(). Destroyed depthmcu_t 0000016364AF6190
[ trace    ] : usbcmd_t_destroy(). Destroyed usbcmd_t 0000016364AECF30
[ trace    ] : colormcu_t_destroy(). Destroyed colormcu_t 0000016364AAA0F0
[ trace    ] : logger_t_destroy(). Destroyed logger_t 0000016364AAA4B0
[ trace    ] : k4a_device_t_destroy(). Destroyed k4a_device_t 0000016364B14A10

wes-b, 2019-07-11:
The problem is this:
```
[ error ] : RuntimeClassInitialize(). Failed to activate source in open camera: 0x80070005
```
0x80070005 = E_ACCESSDENIED For some reason the camera is not accessible. I might consider what other applications on the computer might be opening the camera and changing access permissions.

monkeykat, 2019-07-11:
Okay, I am thinking that my organization is blocking the camera on the computer or something.... I am going to put a ticket in with them to see if they can unblock it. Thank you for your help, I will post once I get with them if it solves the problem or not.

vemeli, 2019-07-16:
If you are using Windows 10, follow this link to allow your apps to access your camera: https://mywindowshub.com/allow-or-deny-apps-to-have-access-to-webcam-in-windows-10/

wes-b, 2019-07-16:
Thanks @guzza100, I confirmed I get the same error when disabling the camera 
```
RuntimeClassInitialize(). Failed to activate source in open camera: 0x80070005
```

closed 2019-07-16 by wes-b

R3adOnlyJoe, 2019-08-20:
After I clicked `Refresh Devices`, I got this errror from the log: 
`RuntimeClassInitialize(). Failed to find color camera in open camera: 0xc00d36d5`. 
Any help?

wes-b, 2019-08-20:
0xc00d36d5 = MF_E_NOT_FOUND; make sure the camera is showing up in device manager, one it shows up there, the camera should work with the in box camera app.

ivoslove, 2020-08-27:
> I am having an issue where the Kinect camera is showing up in the Device Manager, but the Kinect Viewer just says "No available devices." The Kinect Viewer works fine on another computer, so is not defective.
> 
> Windows 10 1809
> USB Host Controller: Intel USB 3.0 eXtensible Host Controller - 1.0

I had the same problem. Did you solve it later? How was it solved?

guzza100, 2020-08-28:
I believe it was an access denied issue specific to my laptop.  So either your firewall or antivirus program is blocking access for security purposes.  Check those settings and allow the Kinect access for camera and microphone purposes.

wangcc123-dave, 2022-08-02:
> 0xc00d36d5 = MF_E_NOT_FOUND; make sure the camera is showing up in device manager, one it shows up there, the camera should work with the in box camera app.

After I clicked Refresh Devices, I got this error from the log:
RuntimeClassInitialize(). Failed to find color camera in open camera: 0x80070003
Any help?

wangcc123-dave, 2022-08-02:
> After I clicked Refresh Devices, I got this error from the log: RuntimeClassInitialize(). Failed to find color camera in open camera: 0x80070003 Any help?

I have opened access to camera and microphone purposes, but it didn't work.

bhadrik-infivr, 2023-09-05:
I am having the same issue but log is different. It is saying device being used but it is not and any other application is able to use this camera. Any Idea how can I solve this.

-----LOG-----
[ trace    ] : k4a_device_t_create(). Created   k4a_device_t 0000014F2EC100B0
[ trace    ] : depthmcu_t_create(). Created   depthmcu_t 0000014F2EBAAC90
[ trace    ] : usbcmd_t_create(). Created   usbcmd_t 0000014F2EC0D250
[ critical ] : find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
[ trace    ] : usbcmd_t_destroy(). Destroyed usbcmd_t 0000014F2EC0D250
[ error    ] : usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[ trace    ] : depthmcu_t_destroy(). Destroyed depthmcu_t 0000014F2EBAAC90
[ error    ] : depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
[ trace    ] : k4a_device_t_destroy(). Destroyed k4a_device_t 0000014F2EC100B0

prasq, 2024-05-15:
Hi There, Greetings

I am new to Azure Kinect Device, It shows steady white light, it shows in my device manager, I can see it in the camera out put in my camera app, but it is not recognized in the “Azure Kinect Viewer” and the Log: “[ trace ] : k4a_device_t_create(). Created k4a_device_t 0000021507027640
[ trace ] : depthmcu_t_create(). Created depthmcu_t 0000021506FA7330
[ trace ] : usbcmd_t_create(). Created usbcmd_t 0000021507018780
[ info ] : find_libusb_device(). Container ID found: {8898924f-7ca3-4e79-6194-aa7adbcf22b1}
[ info ] : populate_serialnumber(). Serial Number found 000341614612
[ trace ] : usb_cmd_io(). XFR: Cmd=00000115, PayloadSize=255
[ trace ] : colormcu_t_create(). Created colormcu_t 0000021506FCFD30
[ trace ] : usbcmd_t_create(). Created usbcmd_t 00000215070193C0
[ error ] : find_libusb_device(). Unable to find Container ID: {8898924f-7ca3-4e79-6194-aa7adbcf22b1}
[ trace ] : usbcmd_t_destroy(). Destroyed usbcmd_t 00000215070193C0
[ error ] : usb_cmd_create(USB_DEVICE_COLOR_IMU_PROCESSOR, NULL_INDEX, container_id, &colormcu->usb_cmd) returned failure in colormcu_create()
[ trace ] : colormcu_t_destroy(). Destroyed colormcu_t 0000021506FCFD30
[ error ] : colormcu_create(container_id, &device->colormcu) returned failure in k4a_device_open()
[ trace ] : usbcmd_t_destroy(). Destroyed usbcmd_t 0000021507018780
[ trace ] : depthmcu_t_destroy(). Destroyed depthmcu_t 0000021506FA7330
[ trace ] : k4a_device_t_destroy(). Destroyed k4a_device_t 0000021507027640
“. how to solve this. and Also kindly help me how to upgrade the firmware, thanks and Regards.