microsoft/Azure-Kinect-Sensor-SDK #476: turbojpeg.h file not found

frenche6, 2019-07-10:
**Describe the bug**
For the transformation project, issue finding the 'turbojpeg.h file.
Actual error is: "Cannot open include file: 'turbojpeg.h': No such file or directory".

When navigating to the extern folder, the libjpeg-turbo folder is empty except for the CMakeLists.txt file.

This may be an issue of me not knowing where the file is located to link it properly.

**To Reproduce**
1. Open the examples solution
2. Rebuild solution
3. See Error

**Expected behavior**

To be able to reference the file 'turbojpeg.h', and remove error.

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS with Version: Windows 1803
 - SDK Version:  1.1.0
 - Firmware version:  Not sure where to find

wes-b, 2019-07-10:
I am guessing you have a fresh clone and have not yet built the SDK. So the git submodules are not populated. You need to run ```git submodule update --init --recursive``` to pulling the header down.

frenche6, 2019-07-10:
Thank you for the suggestion, that led me in the correct direction. I downloaded the project, with submodules this time, and I am able to finally run the cmake successfully.

closed 2019-07-10 by frenche6