microsoft/Azure-Kinect-Sensor-SDK #477: NFOV_BINNED broken depth image displaying on NUC8i7HVK - 1.1.0 SDK

somacoder, 2019-07-10:
NFOV_BINNED displays a broken depth image when running on a NUC8i7HVK in the Kinect Viewer from the SDK 1.1.0 installer in usage:
https://user-images.githubusercontent.com/25708712/60984625-dfacb800-a2f0-11e9-9d56-e7f928cdfba5.PNG

This only occurs on the NUC8i7HVK  and not on my alternate desktop system. It also occurs across multiple different Kinect Hardware units.

This is with all standard connections attached (USB 3.0 and Power).
System: NUC8i7HVK  w/ 32 GB RAM
OS: Windows 10 Version 1903 (OS Build 18362.239)
SDK: 1.1.0

Firmware 
RGB camera: 1.6.98
Depth camera: 1.6.70
Audio: 1.6.14

somacoder, 2019-07-10:
As a followup to this, after testing, this issue does not occur in Ubuntu 18.04 on the NUC8i7HVK.

rabbitdaxi, 2019-07-11:
@somacoder thank you for reporting this issue, we definitely want to know more about this issue. May I ask two questions to help us narrow down a little bit more:
1) It seems this issue only happens on the NUC8i7HVK device on windows, and also this issue does not happen on other desktop windows system. Have you try to update the graphics driver of the NUC8i7HVK see whether it helped addressing this issue? Please share what your current graphics driver version is.
2) On the NUC8i7HVK, does this issue also happen to other 3 depth modes, or it only happens to the NFOV_BINNED?

somacoder, 2019-07-12:
Thanks for the followup. I've since re-imaged to Linux and am programming there (my goal all along once Linux became stable). As such, this system specific issue is less a priority over the Linux bugs currently for me, specifically point cloud visualization bug in the viewer and the libusb multi-camera support bug in the recorder. However, given the NUC8i7HVK will almost certainly be a go to system for this device moving forward, it makes sense to address eventually.

1. I did not update the graphics manually. The RX Vega M is strange in that I think the driver is exclusively distributed by Intel not AMD, so I didn't get around to manually installing that. This could have very well have solved the problem. I can't say what the driver was before I re-imaged.
2. This only occurred with NFOV_BINNED.

I will make it a priority to install Windows again on a separate disk to test again when I get a little more spare time.

wes-b, 2019-07-12:
Are there issues opened for the 2 bugs you mentioned above?

rabbitdaxi, 2019-07-12:
@somacoder thank you for answer my questions. Yes, I think it is definitely worth to try repro this issue on this device and see whether you can update the graphic driver can help solve the issue. We have experienced one Intel Graphics driver issue before results in a color point cloud missing points https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/294, and somehow the windows update on surface does not pull the latest intel graphics driver automatically and manually update fixed the issue. Therefore, please try similar thing on the NUC8i7HVK, likely causing by the graphics driver. 

I will keep an eye on this issue and looking forward to hear your testing results

somacoder, 2019-07-12:
Sounds good! Thanks all. And to @wes-b, yes, those are being handled and look to be resolved soon. I know you all are hard at work and I appreciate the proper triaging!

rabbitdaxi, 2019-07-15:
@somacoder given it may take some time for you to try repro again and testing the update the Intel GPU driver, we are going to close this issue for now, and if update the Intel GPU driver did not fix your issue, please reopen this issue with your repro and testing result at then. Thank you a lot for actively on this issue!

closed 2019-07-15 by rabbitdaxi

somacoder, 2019-07-16:
Updating to the newest driver here did not fix the problem. https://downloadcenter.intel.com/download/28600/Radeon-RX-Vega-M-Graphics-Driver-for-Windows-10-64-bit-for-NUC8i7HNK-NUC8i7HVK?product=126143

somacoder, 2019-07-16:
Success! If you download the Radeon software for the standard RX Vega 64, a driver update will be installed over Intel's advertised driver for this unit, fixing this bug. Strange that Intel communicates having to use their driver available on their website. Doubly strange considering that AMD also doesn't advertise support for the Radeon™ RX Vega M anywhere on their site. https://www.amd.com/en/support/graphics/radeon-rx-vega-series/radeon-rx-vega-series/radeon-rx-vega-64

rabbitdaxi, 2019-07-17:
@somacoder thank you very much for the effort, we really appreciated your follow through here. It is definitely make sense this is some graphic driver bug, and it is also glad to know the newer version that you updated manually did fix the issue. This is very helpful for others if they are hitting same issue.