microsoft/Azure-Kinect-Sensor-SDK #483: what is color space and reference level of RGB camera?  bt.601, bt.709, bt.2020, ...?

diablodale, 2019-07-11:
The colorspace and reference levels (aka range aka swing) of the RBG camera data is missing in all known documentation. The colorspace and its reference white/black must be known and consistently used by Microsoft and developers so that conversions between colorspaces and formats are accurate. Without clarity, errors and color shifts easily occur in core Kinect DK and in customer code. (FYI, I found and assisted Microsoft to fix colorspace code in the Kinect v2 SDK)

## Repo

* Look in your SDK docs, Azure guides, 3rd party articles, or the customer-facing OV12A10 product datasheet.
* Do a internet search for "Azure Kinect colorspace"

## Actual

I do not see it in your SDK docs, Azure guides, 3rd party articles, or even the customer-facing OV12A10 product datasheet.

## Expected

Somewhere to be listed, **for example**

>RGB camera colorspace:  BT.709 full-range (0->255)

That was **only an example** and not the actual.
By knowing this, a developer can take, for example, the YUY2 frames and convert them using custom code into a wide variety of output formats using well-known transformation formula.
This colorspace topic has been well explored by the Kinectv2 team and the XBox team.

diablodale, 2019-07-11:
I have suspicions...that something isn't correct in the DK.
Please reference https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/95f1d95f1f335b57a350a80a3a62e98e1ee4258d/tools/k4aviewer/k4acolorimageconverter.cpp#L104

That line of code is calling libyuv::YUY2ToARGB()
It appears that the DK uses the submodule https://chromium.googlesource.com/libyuv/libyuv
and the code in that submodule for that function is here https://chromium.googlesource.com/libyuv/libyuv/+/refs/heads/master/source/convert_argb.cc#2121
and that library code calls on line 2179...
```
YUY2ToARGBRow(src_yuy2, dst_argb, &kYuvI601Constants, width);
```

That requires that the camera data coming from the KfA camera must be BT.601 and a yet-undetermined range. (I would have to further dig into code to see what range it is).

Given that the RGB camera in KfA was made in 2019 and it has more than HD resolution, it is *very unlikely* the camera delivers BT.601 data. This then suggests that non-BT.601 data is being manipulated/converted using BT.601 algorithms. This leads to inaccurate color shifts.

And still the range is unknown.

rabbitdaxi, 2019-07-15:
@diablodale thank you for reporting this. We are requesting this information from the HW team and will get back to you once we confirmed it.

rabbitdaxi, 2019-07-17:
@diablodale we just get back the confirmation from the color camera hardware team, they confirm the color space conversion standard is BT.601 for this color camera. Therefore, the current k4aviewer is doing the right thing, and we will add this information to the documentation location https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification.

diablodale, 2019-07-17:
That is surprising and a step backward as HD cameras and modern video pipelines use bt.709 or newer.
Please note...the Kinect v2 sensor...was full-range YCbCr **BT.709**.

Technically, it is possible the hardware team designed the hardware and firmware to be BT.601. But it is very very odd. And outdated.

Please do update the documentation as you write and **also** put if it is full-range or studio-range. Listing only BT.601 is not enough.

rabbitdaxi, 2019-07-17:
Based on what we heard from the hardware and firmware team, it should be: BT.601 full-range. We will update the documentation. Thank you for the feedback

rabbitdaxi, 2019-07-18:
We have updated the documentation with adding this information https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification

closed 2019-07-18 by rabbitdaxi