microsoft/Azure-Kinect-Sensor-SDK #487: Firmware Missing in Azure Kinect SDK 1.1.0.msi

jbrownkramer, 2019-07-12:
**Describe the bug**

The [firmware update instructions](https://docs.microsoft.com/en-us/azure/kinect-dk/update-device-firmware) say "In the SDK install location under (SDK install location)\tools\ you should find: ... A Firmware .bin file in the firmware folder, such as AzureKinectDK_Fw_1.5.926614.bin.".  I have installed the SDK by downloading and running the msi file from [https://docs.microsoft.com/en-us/azure/kinect-dk/sensor-sdk-download](https://docs.microsoft.com/en-us/azure/kinect-dk/sensor-sdk-download), and there is no .bin file in the \tools\ directory or anywhere else that I can see.

wes-b, 2019-07-12:
Yes, thank you for the reminder. We need to start adding the firmware. So far there have not been any updates to release.

jbrownkramer, 2019-07-12:
Oh, OK.  We received a camera recently with the following firmware.  I think you're telling me that this is the latest.  Is that right?

```
Current Firmware Versions:
  RGB camera firmware:      1.6.98
  Depth camera firmware:    1.6.70
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft
```

wes-b, 2019-07-12:
Yes, that is correct.

wes-b, 2019-07-17:
Yes, this is correct. 

And with a recent check-in to an internal repro, our next released NuGet and MSI installer will have the firmware. For now Linux users will have to crack one of these open to get the firmware if needed.

closed 2019-07-17 by wes-b

wes-b, 2020-07-13:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#msis