microsoft/Azure-Kinect-Sensor-SDK #492: body tracking viewer error

chankamlam, 2019-07-13:
as the pic blew

[image]

OS:Windows10
GPU:low than 1070
CUDA：10

wes-b, 2019-07-15:
k4abt_simple_3d_viewer.exe

"The application cannot be started normally (Oxc0000142). Click OK to close the application."

wes-b, 2019-07-15:
0xC0000142 = STATUS_DLL_INIT_FAILED

Check that you have all the required dependecies.

cdedmonds, 2019-07-15:
@chankamlam can you confirm that:
1. CUDNN is installed and in your path
2. The VC++ 14 runtime is installed

We are working to create better error messages for missing dependencies in a future release.

chankamlam, 2019-07-16:
How 2 check VC ++ 14 installed or not ,I have been install vs studio 2019, seems it must be already installed

chankamlam, 2019-07-16:
CUDNN is it meant CUDA? I have been installed CUDA 10, but my GPU Display card is low than 1070, is it the key point of the error

yijiew, 2019-07-16:
@chankamlam cuDNN is a separate library than CUDA. It needs to be separately installed. Please follow the instruction here for detailed installation steps: https://docs.microsoft.com/en-us/azure/Kinect-dk/body-sdk-setup. 

The body tracking SDK can run on some graphics card that is lower than 1070. We have tested to run it on GTX 1050Ti and it runs OK. May I ask what graphics card are you using?

chankamlam, 2019-07-17:
@yijiew i have installed CUDA10 before i opened it

chankamlam, 2019-07-17:
@yijiew i had installed CUDA10 before i opened it

yijiew, 2019-07-17:
@chankamlam But did you install cuDNN? As I mentioned, cuDNN needs to be installed separately.

chankamlam, 2019-07-17:
What s cuDNN? official website did not said that needs to installed the cuDNN,how 2 install that?

cdedmonds, 2019-07-17:
@chankamlam please follow the directions here:
https://docs.microsoft.com/en-us/azure/Kinect-dk/body-sdk-setup

venscn, 2019-07-18:
https://developer.nvidia.com/rdp/cudnn-archive

rabbitdaxi, 2019-07-19:
@chankamlam please let us know whether you following the body tracking sdk setup process did fix your issue.

cdedmonds, 2019-07-19:
Closing this issue since this seems to be a case of missing dependencies. @chankamlam please let reactivate if you have more details to share.

closed 2019-07-19 by cdedmonds

twt6, 2019-09-30:
once you download the cudnn file, what do you do with it? Save it somewhere specific? Open it? I'm still getting error 0xc0000142

https://developer.nvidia.com/rdp/cudnn-archive

yijiew, 2019-10-01:
@twt6 : you should copy the cudnn.dll to either the cuda folder or your project folder for it to be loaded correctly.

twt6, 2019-10-01:
Thank you so much for the response! I'm pretty new to all of this. 

I found a file called "cudnn64_7.dll". The path was CUDA > bin > cudnn64_7.dll. Is that the correct file?

I don't know for sure what you mean by cuda folder or project folder. I found this folder and tried it there but it didn't work:
[image]

twt6, 2019-10-01:
@yijiew

I tried it in this folder as well, still didn't work:

[image]

twt6, 2019-10-01:
@yijiew

I just tried this:

https://www.quora.com/How-do-I-register-DLL-files-in-a-Windows-10-64-bit

But it gave me this error:

[image]

yijiew, 2019-10-01:
@twt6 What version of the SDK are you using? If you download the latest SDK (version 0.9.3), the cudnn64_7.dll should be already included in the tools folder and everything should work out of the box.

twt6, 2019-10-03:
I am using 0.9.3. I have uninstalled and reinstalled it, and it still is not opening. We downloaded binaries on this page:
https://www.nuget.org/packages/Microsoft.Azure.Kinect.BodyTracking/0.9.3
and put them in the Bin folder. Still not working.

ianychoi, 2019-10-14:
Thank you for the info - I confirm that using the latest version with dependencies (I cloned some old projects which didn't include them) work very well.

@twt6 If you are using Visual Studio, you can simply remove and re-add the nuget package on *References* menu in your solution explorer. Or please check whether the firmware on the device is up-to-date or not.