microsoft/Azure-Kinect-Sensor-SDK #494: Give a more salient error message when the error is due to depthengine.dll missing

this-username-is-taken, 2019-07-14:
I've ran into a couple situations (often when I migrate a project) where some function in k4a sdk fails, I spend a lot of time debugging, then realize it's because depthengine.dll is not in the bin folder, but there isn't any error message or hint that that is the case.

The latest example of such behavior I ran into is when I tried to create a `k4a_transform_t` with a `k4a_calibration_t`. The function `k4a_transformation_create` simply returns 0 without giving additional information. I spent a lot of time examining if my calibration parameters were incorrect, before realizing it's simply because I didn't have depthengine.dll.

I think a simple solution that could potentially save developers a lot of time is to throw a runtime exception whenever any k4a api gets called with depthengine.dll missing.

rabbitdaxi, 2019-07-15:
@this-username-is-taken thank you for opening this issue, there should be some error message in the console window if you are missing depthengine_1_0.dll, e.g. I just tried the transformation_example.exe that missing the depth engine in the same directory, what I see is:
E:\GIT\Azure-Kinect-Sensor-SDK\build\bin>transformation_example.exe capture d:\
[2019-07-15 14:58:27.813] [error] [t=12456] ..\src\dynlib\dynlib_windows.c (79): dynlib_create(). Failed to load DLL depthengine_1_0 with error code: 126
[2019-07-15 14:58:27.814] [error] [t=12456] ..\src\deloader\deloader.c (70): load_depth_engine(). Failed to Load Depth Engine Plugin (depthengine). Depth functionality will not work
[2019-07-15 14:58:27.816] [error] [t=12456] ..\src\deloader\deloader.c (71): load_depth_engine(). Make sure the depth engine plugin is in your loaders path
[2019-07-15 14:58:27.817] [error] [t=12456] ..\src\deloader\deloader.c (178): deloader_transform_engine_create_and_initialize(). Failed to load depth engine plugin
[2019-07-15 14:58:27.819] [error] [t=12456] ..\src\tewrapper\tewrapper.c (57): transform_engine_start_helper(). Transform engine create and initialize failed with error code: 108.
...

this-username-is-taken, 2019-07-15:
Thanks for looking into this. The error message didn't show up for me. I realized earlier that K4A logging depends on you calling `k4a_device_open`. Since I was working on recordings only with the K4A API, I didn't get any logs. But I suppose this would be a separate issue.

wes-b, 2019-07-15:
Yes, please file an issue on this.

this-username-is-taken, 2019-07-15:
Issue filed here: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/506

Closing this one.

closed 2019-07-15 by this-username-is-taken