microsoft/Azure-Kinect-Sensor-SDK #495: Please clarify Depth Engine Linux OpenGL requirements in docs

jcsoo, 2019-07-14:
The current docs in [Using Azure Kinect SDK](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md) are vague on Linux dependencies, simply listing "OpenGL". The page [Azure Kinect sensor SDK system requirements](https://docs.microsoft.com/en-us/azure/Kinect-dk/system-requirements) on the main site could be a bit more specific - for Linux it lists "Linux Ubuntu 18.04 (x64) with OpenGLv4.4 or later GPU driver", but it's not clear if there is some subset of SDK functionality that works with a lower OpenGL version or if nothing works at all.

In fact it appears that the depth sensor capture code in the Depth Engine is specifically what has the dependency on OpenGL 4.4. The enumeration and firmware examples (including k4aviewer when only streaming from the color camera) all seem to run fine on OpenGL 3.3, but k4arecord fails with this error:

```
Device version: Rel; C: 1.6.98; D: 1.6.70[6109.7]; A: 1.6.14
[2019-07-14 06:45:44.871] [trace] [t=27389] ../src/sdk/k4a.c (704): k4a_device_start_cameras(). k4a_device_start_cameras starting
[2019-07-14 06:45:44.871] [info] [t=27389] ../src/sdk/k4a.c (722): k4a_device_start_cameras(). Starting camera's with the following config.
[2019-07-14 06:45:44.871] [info] [t=27389] ../src/sdk/k4a.c (723): k4a_device_start_cameras().     color_format:0
[2019-07-14 06:45:44.871] [info] [t=27389] ../src/sdk/k4a.c (724): k4a_device_start_cameras().     color_resolution:2
[2019-07-14 06:45:44.871] [info] [t=27389] ../src/sdk/k4a.c (725): k4a_device_start_cameras().     depth_mode:2
[2019-07-14 06:45:44.871] [info] [t=27389] ../src/sdk/k4a.c (726): k4a_device_start_cameras().     camera_fps:2
[2019-07-14 06:45:44.871] [info] [t=27389] ../src/sdk/k4a.c (727): k4a_device_start_cameras().     synchronized_images_only:0
[2019-07-14 06:45:44.871] [info] [t=27389] ../src/sdk/k4a.c (728): k4a_device_start_cameras().     depth_delay_off_color_usec:0
[2019-07-14 06:45:44.871] [info] [t=27389] ../src/sdk/k4a.c (729): k4a_device_start_cameras().     wired_sync_mode:0
[2019-07-14 06:45:44.871] [info] [t=27389] ../src/sdk/k4a.c (730): k4a_device_start_cameras().     subordinate_delay_off_master_usec:0
[2019-07-14 06:45:44.871] [info] [t=27389] ../src/sdk/k4a.c (731): k4a_device_start_cameras().     disable_streaming_indicator:0
[2019-07-14 06:45:44.871] [info] [t=27389] ../src/sdk/k4a.c (737): k4a_device_start_cameras(). colormcu_set_multi_device_mode
[2019-07-14 06:45:44.871] [trace] [t=27389] ../src/usbcommand/usbcommand.c (546): usb_cmd_io(). XFR: Cmd=80000001, CmdLength=13, PayloadSize=0, CmdData=00000000 00000000...
[2019-07-14 06:45:44.881] [info] [t=27389] ../src/sdk/k4a.c (744): k4a_device_start_cameras(). capturesync_start
[2019-07-14 06:45:44.881] [info] [t=27389] ../src/sdk/k4a.c (753): k4a_device_start_cameras(). depth_start
[2019-07-14 06:45:44.881] [trace] [t=27389] ../src/usbcommand/usbcommand.c (546): usb_cmd_io(). XFR: Cmd=000000e1, CmdLength=4, PayloadSize=0, CmdData=00000004 8d535400...
[2019-07-14 06:45:44.885] [trace] [t=27389] ../src/usbcommand/usbcommand.c (546): usb_cmd_io(). XFR: Cmd=00000022, CmdLength=4, PayloadSize=2000000, CmdData=00000002 8d535400...
[2019-07-14 06:45:44.890] [trace] [t=27400] ../src/dynlib/dynlib_linux.c (23): dynlib_t_create(). Created   dynlib_t 0x7f9160000b50
[2019-07-14 06:45:44.892] [info] [t=27400] ../src/deloader/deloader.c (32): verify_plugin(). Loaded K4A Plugin with version: 1.0.7
X Error of failed request:  GLXBadFBConfig
  Major opcode of failed request:  153 (GLX)
  Minor opcode of failed request:  34 ()
  Serial number of failed request:  30
  Current serial number in output stream:  29
```

This is disappointing but not totally unexpected. Unfortunately this prevents usage in VMWare Fusion as well as on headless servers that might otherwise be used as network capture devices.

Could you update the SDK docs to be more specific on the OpenGL version required for depth camera functionality? It might also be useful to have a more obvious error message when the required version is not available.

cdedmonds, 2019-07-15:
Thanks @jcsoo we will look at adding the specific OpenGL version to the documentation.

rabbitdaxi, 2019-07-16:
Also will improve the error message here instead of just showing the OpenGl error here, we explicitly add the ask of checking OpenGL core profile version and suggest update GPU driver

closed 2019-08-01 by rabbitdaxi