microsoft/Azure-Kinect-Sensor-SDK #496: libusb erros that devices is no longer connected!

wqs111000, 2019-07-15:
**Describe the bug**

When  a single Sensor opened, It seems to be working properly. But LibUsb always prints out error messages, as the figure below shows.
**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version: 
[image]

wes-b, 2019-07-15:
Good issue, I have seen this too. Its a delayed message from LibUsb after a device is disconnected.

Brent-A, 2019-07-15:
Related to #485 and #302.

PR #695 (2019-09-03): Error message cleanup
Fixes #567
Fixes #496
Fixes #302

### Description of the changes:
- For access denied errors, where the device was already opened, I added additional logging to trigger the user to think about what might be cause the device to be unavailable. 
- The use of LIBUSB now disables logging when calling libusb_open and libusb_get_device_list as both those API's return errors that can't be disabled.
- Rename 'queue_error' to 'queue_stop' as it is a better captures the state transition and allows de-escalation of error messages.

Errors and Warnings now seen when starting up K4aVi [... cut]
closed 2019-09-03 by wes-b