microsoft/Azure-Kinect-Sensor-SDK #497: Does not work in Color Camera mode

HukoOo, 2019-07-15:
**Describe the bug**

Build is clean and I tried to start camera with k4aviewer.
'Enable Depth Camera' is OK, but 'Enable Color Camera' is not working like below error.

I tested in windows and working well.Not hardware problem.

**To Reproduce**

**Expected behavior**

**Logs**

[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  77816077
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  77849411
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  77882744
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  77916077
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:  77949411
[ warning  ] : depth_engine_thread(). Dropping depth image due to bad timestamp at startup
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     16077 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     49411 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     82744 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    116077 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    149411 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    182744 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    216077 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    249411 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    282744 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    316077 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    349411 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    382744 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    416077 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    449411 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    482744 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    516077 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    549411 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    582744 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    616077 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    649411 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    682744 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    716077 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    749411 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    782744 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    816077 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    849411 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    882744 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    916077 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    949411 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    982744 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1016077 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1049411 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1082744 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1116077 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1149411 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1182744 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1216077 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1249411 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1282744 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1316077 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1349411 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1382744 type:Depth
[ warning  ] : queue_disable(). Waiting for blocking call to complete
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : capturesync_add_capture(). Capture Error Detected, Depth 

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version:  Ubuntu 18.04
 - SDK Version:   2.0.1
 - Firmware version:  

**Additional context**

wes-b, 2019-07-15:
I think your device needs to be power cycled. I don't see an error starting the color camera, just that the data is failing to arrive. Also note #485 that may be impacting you.

wes-b, 2019-07-18:
Any update on this?

closed 2019-07-22 by wes-b