microsoft/Azure-Kinect-Sensor-SDK #498: Unclear documentation for k4a_calibration_2d_to_2d and k4a_calibration_2d_to_3d

this-username-is-taken, 2019-07-15:
Both functions require a depth argument when being called. However, when the source camera type is color, there is no depth information. The documentation doesn't specify what the depth argument should be and what the expected behavior is in such case.

mbleyer, 2019-07-15:
We allow for the color camera to be chosen as the source camera type in order to keep these functions as general as possible. One way to derive the depth value in the color camera geometry is to use the function k4a_transformation_depth_image_to_color_camera().

this-username-is-taken, 2019-07-15:
Thanks Michael. I followed up with Andrew earlier and got clarification on this issue. My only concern is that this is not immediately clear from the documentation. Given that there is no example code on the usage of k4a_calibration_2d_to_2d, I worry that other people may also get confused as I did.

rabbitdaxi, 2019-07-15:
@this-username-is-taken , I think one thing we can do is update the documents in how-to section https://docs.microsoft.com/en-us/azure/Kinect-dk/use-calibration-functions, or also this function's parameter source_depth_mm's description with the additional message "one way to derive the depth value in the color camera geometry is to use the function k4a_transformation_depth_image_to_color_camera()"

PR #512 (2019-07-16): Improve the sensor sdk calibration function usage

## Fixes #498 

### Description of the changes:
- Improve the doc to explain how to get depth value in color camera space when using calibration function

### Before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [x] I [built my changes](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) locally
- [x] I ran the [unit tests](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/testing.md)
- [x] I [... cut]
closed 2019-07-16 by rabbitdaxi